ARM-software/sbsa-acs #13: Timer_wd testcases: Need delay before checking timer status.

pttuan, 2017-06-20:
In test_pool/timer_wd/test_t004.c
``` 
  val_gic_install_isr(intid, isr);

  val_timer_set_phy_el2(timer_expire_val);

  while ((--timeout > 0) && (IS_RESULT_PENDING(val_get_status(index))));
```
After val_timer_set_phy_el2, we need a delay or better solution as in case of CPU running very fast (example: 3Ghz), the timeout value is zero before the timer is elapsed. So the test case will fail in this case.

The same issue is applied for timer_wd/test_t005.c

Hope it helps.

prasanth-pulla, 2017-06-23:
Thank you for pointing this out. In the upcoming release, we will change the timeout and timer expire values to have much larger gap than now.

prasanth-pulla, 2017-06-28:
checkin on 28-jun on master branch - Issue fixed by changing the timeout and timer expiry values in all related tests. Please check and let us know.

closed 2017-06-28 by prasanth-pulla