ARM-software/sbsa-acs #64: Issue with BigEndianCheck function with MMU enabled

jbrasen, 2019-02-25:
I am seeing a crash while executing on system where cpu endianness changes are allowed in the BigEndianCheck function in https://github.com/ARM-software/sbsa-acs/blob/53523295288d4997abf921c60fc5e45d2cc61f88/val/src/AArch64/PeTestSupport.S

Next instruction after the sctlr set is failing. This seems to be due to a translation table lookup error as the EE bit changes endianness for Stage 1 translation table walks as well as data accesses.

If I clear bit 0 of the sctlr during this test as well things seem to work (Probably need to flush x0 as well in case it just in the cache). Not sure if I am missing something in the setup of the test I didn't take into account when debugging this.

prasanth-pulla, 2019-03-01:
Thank you for raising this issue. We are investigating the same.

BhaskaraBudiredla, 2019-03-12:
Your observation is correct @jbrasen. The sequence of steps to dynamically switch between Little/Big- endian worlds is not correct. We will update the test to address this issue.

BhaskaraBudiredla, 2019-04-01:
We fixed the intent of the test case. The same will be available as part of next SBSA release.

prasanth-pulla, 2019-07-12:
Test c008 updated to change the checks.

closed 2019-07-12 by prasanth-pulla