ARM-software/sbsa-acs #98: VHE checks are not consistent

hrw, 2020-10-29:
When I run "sbsa.efi -l 4" to check for SBSA level 4 then there is check does PE supports Virtual Host extensions:
```
 21 : Check for Virtual host extensions 
       Failed on PE -    0 for Level=  4 : Result:  --FAIL-- 1
```

And then there is an information about in test 205:
```
 205 : Check EL2-Virtual timer interrupt 
       v8.1 VHE not supported on this PE : Result:  -SKIPPED- 1
```

But when I check for SBSA level 3 (via 'sbsa.efi -l 3') then PE checks end at 18 (Check for PMBIRQ signal) but check 205 still gives the same output despite fact that level 3 does not need to implement VHE.

hrw, 2020-10-29:
test_pool/timer_wd/test_t005.c says that test 205 is run only for VHE hardware. So it should not start for SBSA level 3 at all.

prasanth-pulla, 2020-10-30:
Hi Marcin - Test 205 is checking the EL2 virtual timer interrupt. We do have a mapping for the interrupt number defined for this in Level-3 of SBSA specification (Table - PPI assignments). So, this test is run for level-3. 

It so happens that this timer has a dependency on VHE. Here is the statement from Arm ARM.

"This register is present only when FEAT_VHE is implemented. Otherwise, direct accesses to CNTHV_CTL_EL2 are UNDEFINED."

Hence, this test is skipped (not FAIL) when VHE is not present. 

(Also, a feature can be present optionally on a system which is targeting a lower level compliance level if even though it is not mandatory at that level. So, a test may be run but would skip rather than fail.)

Hope this clarifies you question.

hrw, 2020-10-30:
Thanks for explanation.

closed 2020-10-30 by hrw