ARM-software/sbsa-acs #99: Add 'Tests skipped' to summary

hrw, 2020-10-30:
I am using SBSA ACS to test how well SBSA reference platform in qemu looks. So I run for levels 3-6 and lot of tests were skipped due to missing features.

But there is no information about amount of skipped tests at the end of run:

```
     Total Tests run  =  102;  Tests Passed  =   44  Tests Failed =   29
```

So at the end results table looks weird:

SBSA level | tests run | passed | failed
-|-|-|-
  3 |     45    |   32   |    6
  4 |     87    |   44   |   13
  5 |     92    |   44   |   20
  6 |    102    |   44   |   29

prasanth-pulla, 2020-10-30:
Though it is trivial to add skipped to the summary, It is not present on purpose. This is to simply the messaging.

Skipped tests are "if present" statements in the specification. To achieve a compliance level, the focus and requirement is to bring the Failed tests to zero. You can still have multiple skipped tests and still be compliant to a level as long as the fail count is zero. Hence, the focus on Pass and fail only.

Of course, skipped tests = Total - (pass + fail).

hrw, 2020-10-30:
Thanks for explanation.

closed 2020-10-30 by hrw

prasanth-pulla, 2020-10-30:
In general, an SOC vendor know the target level to which they are building their SOC to. I believe the target for QEMU SBSA platform could be level-3 and just look at the 6 failures.

hrw, 2020-10-30:
The goal is to first get level 3 and then follow up.

```
 SBSA Architecture Compliance Suite 
    Version 3.0  

 Starting tests for level  3 (Print level is  3)

 Creating Platform Information Tables 
 PE_INFO: Number of PE detected       :    4 
 GIC_INFO: Number of GICD             :    1 
 GIC_INFO: Number of ITS              :    0 
 GIC ITS Initialization Failed.
 LPI Interrupt related test may not pass.
 TIMER_INFO: Number of system timers  :    0 
 WATCHDOG_INFO: Number of Watchdogs   :    0 
 PCIE_INFO: Number of ECAM regions    :    2 
 PCIE_INFO: Number of RCiEP           :    4 
 PCIE_INFO: Number of RCEC            :    0 
 PCIE_INFO: Number of EP              :    0 
 PCIE_INFO: Number of RP              :    0 
 PCIE_INFO: Number of iEP_EP          :    0 
 PCIE_INFO: Number of iEP_RP          :    0 
 SMMU_INFO: Number of SMMU CTRL       :    0 
 Peripheral: Num of USB controllers   :    1 
 Peripheral: Num of SATA controllers  :    1 
 Peripheral: Num of UART controllers  :    1 

      ***  Starting PE tests ***  
   1 : Check for number of PE            : Result:  PASS 
   2 : Check for SIMD extensions         : Result:  PASS 
   3 : Check for 16-bit ASID support     : Result:  PASS 
   4 : Check PE Granule Support          : Result:  PASS 
   5 : Check Cache Architecture          : Result:  PASS 
   6 : Check HW Coherence support        : Result:  PASS 
   7 : Check Cryptographic extensions    : Result:  PASS 
   8 : Check Little Endian support       : Result:  PASS 
   9 : Check EL2 implementation          : Result:  PASS 
  10 : Check AARCH64 implementation      : Result:  PASS 
  11 : Check PMU Overflow signal         : Result:  PASS 
  12 : Check number of PMU counters      
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 
  13 : Check Synchronous Watchpoints     : Result:  PASS 
  14 : Check number of Breakpoints       : Result:  PASS 
  15 : Check Arch symmetry across PE     : Result:  PASS 
  16 : Check EL3 implementation          : Result:  PASS 
  17 : Check CRC32 instruction support   : Result:  PASS 
  18 : Check for PMBIRQ signal           
       SPE not supported on this PE      : Result:  -SKIPPED- 1 

      *** One or more PE tests have failed... *** 

      ***  Starting GIC tests ***  
 101 : Check GIC version                 : Result:  PASS 
 102 : If PCIe, then GIC implements ITS  
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 
 103 : GIC number of Security states(2)  : Result:  PASS 
 104 : GIC Maintenance Interrupt         
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 

      One or more GIC tests failed. Check Log 

      *** Starting Timer tests ***  
 201 : Check Counter Frequency           : Result:  PASS 
 202 : Check EL0-Phy timer interrupt     : Result:  PASS 
 203 : Check EL0-Virtual timer interrupt : Result:  PASS 
 204 : Check EL2-phy timer interrupt     : Result:  PASS 
 205 : Check EL2-Virtual timer interrupt 
       v8.1 VHE not supported on this PE : Result:  -SKIPPED- 1 
 206 : SYS Timer if PE Timer not ON      
       PE Timers are not always-on.       
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 
 207 : CNTCTLBase & CNTBaseN access      
       No System timers are defined      : Result:  -SKIPPED- 1 

     *** Skipping remaining System timer tests *** 

      *** One or more tests have Failed/Skipped.*** 

      *** Starting Watchdog tests ***  
 301 : Check NS Watchdog Accessibility   
       No Watchdogs reported          0  
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 
 302 : Check Watchdog WS0 interrupt      
       No Watchdogs reported          0  
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 
 303 : Check NS Watchdog Revision        : Result:  -SKIPPED- 1 

      ***One or more tests have failed... *** 

      *** Starting PCIe tests ***  
 401 : Check ECAM Presence               : Result:  PASS 
 402 : Check ECAM value in MCFG table    : Result:  PASS 
 403 : Check ECAM Memory accessibility   : Result:  PASS 

      *** Starting Power and Wakeup semantic tests ***  
       TEST Wakeup from Power Semantic B  
 501 : Wake from EL0 PHY Timer Interrupt : Result:  PASS 
 502 : Wake from EL0 VIRT Timer Interrupt: Result:  PASS 
 503 : Wake from EL2 PHY Timer Interrupt : Result:  PASS 
 504 : Wake from Watchdog WS0 Interrupt  
       No watchdog implemented           : Result:  -SKIPPED- 1 
 505 : Wake from System Timer Interrupt  
       No system timers implemented      : Result:  -SKIPPED- 1 

      *** Starting Peripheral tests ***  
 601 : Check USB CTRL Interface via PCIe : Result:  PASS 
 602 : Check SATA CTRL Interface via PCIe
       Read PCIe_CFG: ECAM Base is zero 
       WARN: SATA CTRL ECAM access failed 0  
       Re-checking SATA CTRL using PciIo protocol       : Result:  PASS 
 603 : Check SBSA UART register offsets  : Result:  PASS 
 604 : Check GENERIC UART Interrupt      
       Test Message                      : Result:  PASS 
 605 : Memory Access to Un-Populated addr: Result:  PASS 

      *** Starting IO Virtualization tests ***  

     No SMMU Controller Found, Skipping SMMU tests...

      *** Starting PCIe Exerciser tests ***  
Exerciser Sbsa compliance is only from Level 4 

     ------------------------------------------------------- 
     Total Tests run  =   45;  Tests Passed  =   33  Tests Failed =    6
     --------------------------------------------------------- 

      *** SBSA tests complete. Reset the system. *** 

```

prasanth-pulla, 2020-10-30:
Looks like PMU, Watchdog and GIC Maintenance interrupt support would need the required support to be added in Qemu.

hrw, 2020-10-30:
Watchdog is merged into qemu. EDK2 support for it is on a way.

hrw, 2020-10-30:
https://snapshots.linaro.org/ldcg/sbsa-acs/4/ has results for all levels if someone is interested.