ARM-software/sbsa-acs #100: Some issues with the latest sbsa 3.0

tphan-ampere, 2020-11-04:
1. Wrong GICR base address
- In APIC table, it can has more than one GICR base address (multiple sockets). The code to find the GICR of current CPU needs to be fixed:
@@ -279,7 +279,10 @@ pal_gic_its_configure (
     if (g_gic_entry->type == ENTRY_TYPE_GICD) {
         g_gic_its_info->GicDBase = g_gic_entry->base;
     } else if (g_gic_entry->type == ENTRY_TYPE_GICRD) {
-        mGicRedistributorBase = g_gic_entry->base;
+        /* Get Redistributor base for Current CPU. */
+        if (!mGicRedistributorBase && GetCurrentCpuRDBase(g_gic_entry->base)) {
+          mGicRedistributorBase = g_gic_entry->base;
+        }
     } else if (g_gic_entry->type == ENTRY_TYPE_GICITS) {
         g_gic_its_info->GicIts[g_gic_its_info->GicNumIts].Base = g_gic_entry->base;
         g_gic_its_info->GicIts[g_gic_its_info->GicNumIts++].ID = g_gic_entry->its_id;
@@ -289,8 +292,7 @@ pal_gic_its_configure (
 
   sbsa_print(AVS_PRINT_INFO, L"GIC Distributor base %x \n", g_gic_its_info->GicDBase);
 
-  /* Get Redistributor base for Current CPU. */
-  g_gic_its_info->GicRdBase = GetCurrentCpuRDBase(mGicRedistributorBase);
+  g_gic_its_info->GicRdBase = mGicRedistributorBase;
   if (g_gic_its_info->GicRdBase == 0) {
     sbsa_print(AVS_PRINT_DEBUG, L"Could not get GIC RD Base.\n", 0);
     return 0xFFFFFFFF;

2. PCIe testcase 448:
- The code to check if read value is valid or not need to compare with the old value, not just based on 0xffffffff
diff --git a/test_pool/pcie/test_p048.c b/test_pool/pcie/test_p048.c
index 62a48d0..59a4dca 100644
--- a/test_pool/pcie/test_p048.c
+++ b/test_pool/pcie/test_p048.c
@@ -52,7 +52,7 @@ payload(void)
   uint32_t dp_type;
   uint32_t pe_index;
   uint32_t tbl_index;
-  uint32_t read_value;
+  uint32_t read_value, old_value;
   uint32_t test_skip = 1;
   uint64_t mem_base;
   uint64_t mem_lim;
@@ -112,6 +112,7 @@ payload(void)
          * Base + 0x10 will always be in the range.
          * Read the same
         */
+        old_value = (*(volatile uint32_t *)(mem_base + MEM_OFFSET_10));
         *(volatile uint32_t *)(mem_base + MEM_OFFSET_10) = KNOWN_DATA;
         read_value = (*(volatile uint32_t *)(mem_base + MEM_OFFSET_10));
 
@@ -124,8 +125,7 @@ exception_return:
           val_pcie_clear_urd(bdf);
           return;
         }
-
-        if ((read_value == PCIE_UNKNOWN_RESPONSE) || val_pcie_is_urd(bdf)) {
+        if ((old_value != read_value && read_value == PCIE_UNKNOWN_RESPONSE) || val_pcie_is_urd(bdf)) {
           val_set_status(pe_index, RESULT_FAIL(g_sbsa_level, TEST_NUM, 02));
           val_pcie_clear_urd(bdf);
           return;

3. Testcase 605:
- I don't know why this test case was modified. The old code works fine to me. The usage of val_memory_get_info to check if an address populated is not correct because it uses UEFI get memory map function which doesn't return address belonged to device address. The correct code is using GetMemorySpaceMap that pal_memory_get_unpopulated_addr uses.
diff --git a/test_pool/peripherals/test_m001.c b/test_pool/peripherals/test_m001.c
index a8536ae..c96c140 100755
--- a/test_pool/peripherals/test_m001.c
+++ b/test_pool/peripherals/test_m001.c
@@ -25,7 +25,6 @@
 #define TEST_DESC  "Memory Access to Un-Populated addr"
 
 #define LOOP_VAR   3          /* Number of Addresses to check */
-#define STEP_SIZE  0x1000000  /* Step size to increment address */
 
 static void *branch_to_test;
 
@@ -51,10 +50,8 @@ void
 payload()
 {
   addr_t   addr;
-  uint64_t attr;
   uint32_t instance = 0;
   uint64_t status;
-  uint32_t loop_var = LOOP_VAR;
   uint32_t index = val_pe_get_index_mpid(val_pe_get_mpid());
 
   val_pe_install_esr(EXCEPT_AARCH64_SYNCHRONOUS_EXCEPTIONS, esr);
@@ -62,31 +59,27 @@ payload()
   /* If we don't find a single un-populated address, mark this test as skipped */
   val_set_status(index, RESULT_SKIP(g_sbsa_level, TEST_NUM, 01));
 
-  /* Get the base address of unpopulated region */
-  status = val_memory_get_unpopulated_addr(&addr, instance);
-  if (status) {
-      val_print(AVS_PRINT_ERR, "\n      Error in obtaining unpopulated memory for instance 0x%d", instance);
-      return;
-  }
-
-  while (loop_var > 0) {
-      if (val_memory_get_info(addr, &attr) == MEM_TYPE_NOT_POPULATED) {
-         /* default value of FAIL, Pass is set in the exception handler */
-          val_set_status(index, RESULT_FAIL(g_sbsa_level, TEST_NUM, 01));
-
-          branch_to_test = &&exception_taken;
-
-          *((volatile uint64_t*)addr) = 0x100;
-exception_taken:
-          /* if the access did not go to our exception handler, fail and exit */
-          if (IS_TEST_FAIL(val_get_status(index))) {
-              val_print(AVS_PRINT_ERR, "\n      Memory access check fails at address = 0x%llx ", addr);
-              return;
-          }
-
-      }
-      addr += STEP_SIZE;
-      loop_var--;
+  while (instance < LOOP_VAR) {
+        /* Get the base address of unpopulated region */
+        status = val_memory_get_unpopulated_addr(&addr, instance);
+        if (status) {
+            val_print(AVS_PRINT_ERR, "\n      Error in obtaining unpopulated memory for instance 0x%d", instance);
+            return;
+        }
+
+        /* default value of FAIL, Pass is set in the exception handler */
+        val_set_status(index, RESULT_FAIL(g_sbsa_level, TEST_NUM, 01));
+
+        branch_to_test = &&exception_taken;
+
+        *((volatile uint64_t*)addr) = 0x100;
+        exception_taken:
+        /* if the access did not go to our exception handler, fail and exit */
+        if (IS_TEST_FAIL(val_get_status(index))) {
+            val_print(AVS_PRINT_ERR, "\n      Memory access check fails at address = 0x%llx ", addr);
+            return;
+        }
+        instance++;
   }
 
 }
diff --git a/platform/pal_uefi/src/pal_peripherals.c b/platform/pal_uefi/src/pal_peripherals.c
index 1936d19..4cbcb6c 100644
--- a/platform/pal_uefi/src/pal_peripherals.c
+++ b/platform/pal_uefi/src/pal_peripherals.c
@@ -321,11 +321,11 @@ pal_memory_get_unpopulated_addr(UINT64 *addr, UINT32 instance)
     {
       if (Memory_instance == instance)
       {
-        if (*addr == 0)
+        if (addr == 0)
           continue;
 
         *addr = MemorySpaceMap->BaseAddress;

edhay, 2020-11-05:
Hi Tuan, 

     We are checking the issues. 
Wil further update.

-- Enterprise ACS Support

edhay, 2020-12-24:
Hi @tphan-ampere,

    The solution for above three issues have been up-streamed.
Could you please verify on your system and confirm.

-- Enterprise ACS Support

closed 2021-01-20 by gowthamsiddarthd