ARM-software/sbsa-acs #102: Synchronous Exception?

hrw, 2020-11-10:
I do wonder is it EDK2 code quality or SBSA-ACS code quality. Or maybe direction of wind in Arctic is at fault...

Most of systems I try to run SBSA ACS behave similar to this:

```
 SBSA Architecture Compliance Suite 
    Version 3.0  

 Starting tests for level  3 (Print level is  2)

 Creating Platform Information Tables 
MPIDR 0 PE num 0 
MPIDR 100 PE num 1 
 PE_INFO: Number of PE detected       :    2 
 GIC_INFO: Number of GICD             :    1 
 GIC_INFO: Number of ITS              :    1 
 TIMER_INFO: Number of system timers  :    0 
 WATCHDOG_INFO: Number of Watchdogs   :    0 
 PCIE_INFO: Number of ECAM regions    :    1 

Synchronous Exception at 0x000000006BF2AAA8
```

It starts and at some point all I get is "Synchronous Exception" and done.

edhay, 2020-11-10:
Hi Marcin,

Could you please run the Sbsa.efi with the below options and send us the logs:

sbsa.efi -v 1 -l 3 

The above options will execute the SBSA tests at Level 3 and also enable the debug logs which will help us narrow down on the issue.

-- ACS Support

hrw, 2020-11-10:
That system was t4g.micro AWS instance. Getting logs from it is a problem as only last 64KB is available.

```
       Calculated config address is 26008000 pal_mmio_read Address = 26008000  Data = 82501D0F 
       Calculated config address is 26008008 pal_mmio_read Address = 26008008  Data = 7000300 
       Calculated config address is 26008034 pal_mmio_read Address = 26008034  Data = 0 
       Calculated config address is 26009000 pal_mmio_read Address = 26009000  Data = 82501D0F 
       Calculated config address is 26009008 pal_mmio_read Address = 26009008  Data = 7000300 
       Calculated config address is 26009034 pal_mmio_read Address = 26009034  Data = 0 
       Calculated config address is 2600A000 pal_mmio_read Address = 2600A000  Data = 82501D0F 
       Calculated config address is 2600A008 pal_mmio_read Address = 2600A008  Data = 7000300 
       Calculated config address is 2600A034 pal_mmio_read Address = 2600A034  Data = 0 
       Calculated config address is 2600B000 pal_mmio_read Address = 2600B000  Data = 82501D0F 
       Calculated config address is 2600B008 pal_mmio_read Address = 2600B008  Data = 7000300 
       Calculated config address is 2600B034 pal_mmio_read Address = 2600B034  Data = 0 
       Calculated config address is 2600C000 pal_mmio_read Address = 2600C000  Data = 82501D0F 
       Calculated config address is 2600C008 pal_mmio_read Address = 2600C008  Data = 7000300 
       Calculated config address is 2600C034 pal_mmio_read Address = 2600C034  Data = 0 
       Calculated config address is 2600D000 pal_mmio_read Address = 2600D000  Data = 82501D0F 
       Calculated config address is 2600D008 pal_mmio_read Address = 2600D008  Data = 7000300 
       Calculated config address is 2600D034 pal_mmio_read Address = 2600D034  Data = 0 
       Calculated config address is 2600E000 pal_mmio_read Address = 2600E000  Data = 82501D0F 
       Calculated config address is 2600E008 pal_mmio_read Address = 2600E008  Data = 7000300 
       Calculated config address is 2600E034 pal_mmio_read Address = 2600E034  Data = 0 
       Calculated config address is 2600F000 pal_mmio_read Address = 2600F000  Data = 82501D0F 
       Calculated config address is 2600F008 pal_mmio_read Address = 2600F008  Data = 7000300 
       Calculated config address is 2600F034 pal_mmio_read Address = 2600F034  Data = 0 
       Calculated config address is 26010000 pal_mmio_read Address = 26010000  Data = FFFFFFFF 
       Calculated config address is 26018000 pal_mmio_read Address = 26018000  Data = FFFFFFFF 
       Calculated config address is 26020000 pal_mmio_read Address = 26020000  Data = 80611D0F 
       Calculated config address is 26020008 pal_mmio_read Address = 26020008  Data = 1080200 
       Calculated config address is 26020034 pal_mmio_read Address = 26020034  Data = 70 
       Calculated config address is 26020070 pal_mmio_read Address = 26020070  Data = 92B010 
       Calculated config address is 26021000 pal_mmio_read Address = 26021000  Data = 80611D0F 
       Calculated config address is 26021008 pal_mmio_read Address = 26021008  Data = 1080200 
       Calculated config address is 26021034 pal_mmio_read Address = 26021034  Data = 70 
       Calculated config address is 26021070 pal_mmio_read Address = 26021070  Data = 92B010 
       Calculated config address is 60040126022000
Synchronous Exception at 0x000000006BF2AAA8
```

Other hardware I have access to are covered by misc NDA documents so I may not be allowed to paste results ;(

edhay, 2020-11-11:
Hi Marcin,

Thank you for the logs, they were helpful.
From the logs prints, we suspect the exception is raised during config space read of PCIE device with BDF(0, 4, 2).
Config space read for pcie devices prior to this one are successful.

We need to further check if the BDF(0, 4, 2) is valid device. Can you please run below commands on uefi shell on the same device and share us the output.

$pci                      // it will print edk2 view of valid pcie devices in system
$pci 00 04 01     //  it will give config space dump for bdf (0, 4, 1)  as in logs config space access for this device were success.
$pci 00 04 02    

--ACS Support

hrw, 2020-11-12:
Sorry but AWS instances are pain when it comes to playing with UEFI shell.

I tried to find easier to debug platform here. Have two options: QEMU 'virt' machine and APM Mustang. Both are not SBSA level 3 and both have SBSA ACS failing in some ways ;D

I do wonder do you have any test suite for SBSA ACS tests or you run it on systems and expect them to be 100% correct with your assumptions?

hrw, 2020-11-14:
Ok, I moved to "qemu-system-aarch64 -M virt" (on x86-64 host) as it is easy to repeat.

```
Shell>fs0:sbsa.efi -v 1 -l 3

 SBSA Architecture Compliance Suite 
    Version 3.0  

 Starting tests for level  3 (Print level is  3)

 Creating Platform Information Tables 
 PE_INFO: Number of PE detected       :    2
 MADT is at FABE0000 and length is F4
GIC DIS base 8000000 
GIC CPUIF base 8010000 
GIC CPUIF base 8010000 
 GIC_INFO: Number of GICD             :    1  
 GIC_INFO: Number of ITS              :    0 
GIC Distributor base 8000000 

Synchronous Exception at 0x00000000FAAB85A8
```

Looked at code. Added several print() calls. If failes in platform/pal_uefi/src_gic_its/sbsa_gic_its.c file on this loop:

```
  for (Index = 0; Index < num_pe; Index++) {
    Affinity = MmioRead32(GicCpuRedistributorBase + ARM_GICR_TYPER + NEXT_DW_OFFSET);
    if (Affinity == CpuAffinity) {
      return GicCpuRedistributorBase;
    }
```

When I removed call to configureGicIts() then tests went further:

```
 SBSA Architecture Compliance Suite 
    Version 3.0  

 Starting tests for level  3 (Print level is  3)

 Creating Platform Information Tables 
 PE_INFO: Number of PE detected       :    2 
 GIC_INFO: Number of GICD             :    1 
 GIC_INFO: Number of ITS              :    0 
 TIMER_INFO: Number of system timers  :    0 
 WATCHDOG_INFO: Number of Watchdogs   :    0 
 PCIE_INFO: Number of ECAM regions    :    1 
 PCIE_INFO: Number of RCiEP           :    0 
 PCIE_INFO: Number of RCEC            :    0 
 PCIE_INFO: Number of EP              :    8 
 PCIE_INFO: Number of RP              :   11 
 PCIE_INFO: Number of iEP_EP          :    0 
 PCIE_INFO: Number of iEP_RP          :    0 
 SMMU_INFO: Number of SMMU CTRL       :    0 
 Peripheral: Num of USB controllers   :    1 
 Peripheral: Num of SATA controllers  :    1 
 Peripheral: Num of UART controllers  :    1 

      ***  Starting PE tests ***  
   1 : Check for number of PE            : Result:  PASS 
   2 : Check for SIMD extensions         
        Unexpected exception occured
        FAR reported = 0x0
        ESR reported = 0x2000000
     ------------------------------------------------------- 
     Total Tests run  =    2;  Tests Passed  =    1  Tests Failed =    0
     --------------------------------------------------------- 

      *** SBSA tests complete. Reset the system. *** 

```

edhay, 2020-11-17:
Hi Marcin,

    Thank you for the logs and the analysis.

After further analysis, we find that the issue with the loop:

  for (Index = 0; Index < num_pe; Index++) {
    Affinity = MmioRead32(GicCpuRedistributorBase + ARM_GICR_TYPER + NEXT_DW_OFFSET);
    if (Affinity == CpuAffinity) {
      return GicCpuRedistributorBase;
    }

is an impact of a known issue. The solution for the the same is in progress and under verification.
We will upstream the fix within a weeks time.

--ACS Support

rajatgoyal47, 2020-12-22:
Hi Marcin,

Fix for the above issue is merged now as part of #113.

Regards,
Rajat

--ACS Support

closed 2021-01-20 by gowthamsiddarthd