ARM-software/sbsa-acs #110: Questions on PCIE test 019

Jingshan-cwm, 2020-12-15:
Multifunction devices must implement minimal ACS features if P2P supported

1.	Codes seems if the device is multi-function device, the device test will be skipped, 
	            if (val_pcie_multifunction_support(dev_bdf))
	                continue;

2.	When acs_data is zero, escape all the tests here, 
        in condition of multi device to be tested, if we can skill the failure device and continue to test the reset of devices.
            val_pcie_read_ext_cap_word(dev_bdf, PCI_EXT_CAPID_ACS, PCI_CAPID_ACS, &acs_data);
            if (!acs_data) {
                val_print(AVS_PRINT_WARN, "\n       ACS feature not supported", 0);
                return;
            }

gowthamsiddarthd, 2020-12-15:
Hi @Jingshan-cwm,

Thank you for reporting this issue.
We are working on providing the fix for this issue.

Regards,
Gowtham
Enterprise ACS Support

gowthamsiddarthd, 2020-12-21:
Hi @Jingshan-cwm,

A fix for the issue and suggestion provided has been pushed as part of https://github.com/ARM-software/sbsa-acs/pull/112

Regards,
Gowtham

closed 2020-12-22 by gowthamsiddarthd