ARM-software/sbsa-acs #118: sbsa pe c022_entry

tingwanyang, 2021-01-18:
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/test_pool/pe/test_c022.c#L29

When running sbsa pe c022_entry item, it is failed. 
This case check ID_AA64ISAR1_EL1 register, sbsa testcase check APA, bits [7:4] must be value 0x1,
but armv8 spec says it maybe other value according to armv8 version. Could you give me some help?

gowthamsiddarthd, 2021-01-19:
Hi Tingwan,

Thank you for reporting this.
The below tests are from the SBSA specification scenario
1.	For Level 4
If PEs implement Armv8.3 pointer signing, the PEs must provide the standard algorithm defined by the Arm architecture [3], identified by ID_AA64ISAR1_EL1[7:4]==0001.
 
2.	For Level 5 and above
All PEs must implement pointer signing using the standard algorithm defined by the Arm architecture [3], as indicated by ID_AA64ISAR1_EL1[7:4]==0001.
 
Yes, you are right. The SBSA Specification and test need an update with the updated register description according to Arm ARM. We will be updating the test in the next ACS release based on the update of the SBSA specification.

The proposed errata for SBSA spec is this – 
All PEs must implement pointer signing using the standard algorithm defined by the Arm architecture [3], as indicated by ID_AA64ISAR1_EL1.APA having a non-zero value.

Regards,
Gowtham

rajatgoyal47, 2021-02-01:
Hi @tingwanyang,

Fix for this issue has been merged as part of #127.
test_c022 is updated to check for non-zero value of ID_AA64ISAR1_EL1[7:4].

Regards,
Rajat

closed 2021-02-01 by rajatgoyal47