ARM-software/sbsa-acs #119: ≤

tingwanyang, 2021-01-18:
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/val/src/AArch64/PeRegSysSupport.S#L372
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/val/src/AArch64/PeRegSysSupport.S#L377
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/val/src/AArch64/PeRegSysSupport.S#L382
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/val/src/AArch64/PeRegSysSupport.S#L387
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/val/src/AArch64/PeRegSysSupport.S#L392
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/val/src/AArch64/PeTestSupport.S#L54
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/val/src/AArch64/PeTestSupport.S#L56
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/val/src/AArch64/PeTestSupport.S#L58
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/val/src/AArch64/PeTestSupport.S#L61
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/val/src/AArch64/PeTestSupport.S#L63

testcase use sign // disable these key instructions, May I  know the reason?

rajatgoyal47, 2021-01-19:
Hi Tingwan,

Thank you for reporting this.
This test checks that PMBIRQ is wired to PPI number 21, we generate the pmbirq in this test.
 
As part of the test,  there is an access to MDCR_EL3 register to set Non-secure Profiling Buffer and we do it using the secure SMC Call.

We will be removing the secure tests from the SBSA, and this test_c018 will be removed as part of this.
Also, we will change the above functions to call the commented code.

Thanks,
Rajat

rajatgoyal47, 2021-02-01:
Hi @tingwanyang,

We have removed this test from the ACS repository as secure side rules are optional in the spec and this test uses secure side functionality. This is not required for SystemReady.
Closing this issue as this change has been merged as part of #125 

Regards,
Rajat

closed 2021-02-01 by rajatgoyal47