ARM-software/sbsa-acs #120: 606 Non secure Access to Secure addr

tingwanyang, 2021-01-18:
https://github.com/ARM-software/sbsa-acs/blob/master/test_pool/peripherals/test_m002.c

when we running this case, it seems make stack destroy, 
the value of the local variable is destroyed.

gowthamsiddarthd, 2021-01-28:
Hi @tingwanyang,

We have removed this test from the ACS repository as secure side rules are optional in the spec and this is not required for SystemReady.
This Github Pull Request has the change and will be merged soon. https://github.com/ARM-software/sbsa-acs/pull/125

Regards,
Gowtham

rajatgoyal47, 2021-02-01:
Hi @tingwanyang,

We have removed this test from the ACS repository as secure side rules are optional in the spec and this is not required for SystemReady.
Closing this issue as this change has been merged as part of #125 

Regards,
Rajat

closed 2021-02-01 by rajatgoyal47