ARM-software/sbsa-acs #121: 902 Check System Genric Counter

tingwanyang, 2021-01-18:
https://github.com/ARM-software/sbsa-acs/blob/a8b46b94b69878247d54823f4dc3713a19b0377e/platform/secure_sw/arm-tf/sbsa_avs_main.c#L116

cntcvl register write and read Comply with the fellowing spec.

[image]

gowthamsiddarthd, 2021-01-28:
Hi @tingwanyang,

The write to CNTCVL does not comply with the spec 'coresight_soc600_technical_reference_manual_100806_0302_00_en.pdf'.
We have removed this test from the ACS repository as secure side rules are optional in the spec and this is not required for SystemReady.
This Github Pull Request has the change and will be merged soon. https://github.com/ARM-software/sbsa-acs/pull/125

Regards,
Gowtham

rajatgoyal47, 2021-02-01:
Hi @tingwanyang,

We have removed this test from the ACS repository as secure side rules are optional in the spec and this is not required for SystemReady.
Closing this issue as this change has been merged as part of #125 

Regards,
Rajat

closed 2021-02-01 by rajatgoyal47