ARM-software/sbsa-acs #124: Questions on PCIE test_p014

YinXuan-png, 2021-01-22:
Hi,
    I had met two problems when testing PCIe test_p0014, as follows:
    1. When calculating pdg, the function pal_dma_mem_get_attrs in sbsa-acs-drv/files/platform/pal_linux/files/src/pal_dma.c uses ttbr1_el1 to calculate pdg. The calculated pgd kernel virtual address is an abnormal address(such as 0x0001_7000_000a_b1000),
 and it is neither in the user Space, no longer kernel space. Accessing it will cause oops in Linux kernel.But using init_mm.pgd will not cause the above problems.
   2. Another problem is the function is_pte which judges pte. According to the meaning of the bit provided by the kernel, it is found that the is_pte interface has a problem.
   My patch can solve two problems above, Please check if there is a problem with it, thanks.
   My patch as follows:
--- a/sbsa-acs-drv/files/platform/pal_linux/files/src/pal_dma.c
+++ b/sbsa-acs-drv/files/platform/pal_linux/files/src/pal_dma.c
@@ -198,7 +201,8 @@ pal_dma_start_to_device(void *dma_source_buf, unsigned int length,
 static int
 is_pte(uint64_t val)
 {
-    return !(val & 0x2);
+    return (val & 0x2);
 }

 /* Decode memory attribute and shareabilty from page table descriptor val*/
@@ -218,6 +222,7 @@ pal_dma_mem_get_attrs(void *buf, uint32_t *attr, uint32_t *sh)
     pud_t *pud;
     pmd_t *pmd;
     pte_t *pte;
     pgd_t *pgd, *swapper_pgd;
     struct page *pg = phys_to_page(read_sysreg(ttbr1_el1));

@@ -225,7 +230,10 @@ pal_dma_mem_get_attrs(void *buf, uint32_t *attr, uint32_t *sh)
     if(!swapper_pgd)
         return -1;

-    pgd = pgd_offset_raw(swapper_pgd, (uint64_t)buf);
+    pgd = pgd_offset_k((uint64_t)buf);
+
     if(!pgd)
         return -1;
     kunmap(pg);
@@ -256,4 +264,43 @@ pal_dma_mem_get_attrs(void *buf, uint32_t *attr, uint32_t *sh)
     }

     return -1;

gowthamsiddarthd, 2021-02-10:
Hi @YinXuan-png,

Thank you for providing the patch. We are validating these changes and will provide you an update shortly.

Regards,
Gowtham Siddarth
Enterprise ACS support team

Shaolin-Xie, 2021-02-15:
Hey Gowtham, any updates on this issue?

vineethrp, 2021-08-12:
@Shaolin-Xie , init_mm is not exported right? How did you get it to work with sbsa_acs_drv?

chetan-rathore, 2022-06-22:
Thanks @YinXuan-png for the patch and apologies for the late reply.

We are checking this test on the recent SR ACS release and will share updates soon.

Thanks,
ACS

chetan-rathore, 2022-06-28:
Hi @Shaolin-Xie  and @YinXuan-png, 

We have raised the required changes at https://gitlab.arm.com/linux-arm/linux-acs/-/merge_requests/18

Thanks,
ACS

closed 2022-06-30 by chetan-rathore

sunnywang-arm, 2022-09-15:
Fix is in https://gitlab.arm.com/linux-arm/linux-acs/-/commit/7add1bb4cf66a83d9ca07d1fcd2a216120fb7ec8