ARM-software/sbsa-acs #126: Questions on PCIE test_p013

YinXuan-png, 2021-01-27:
Hi,
    I had met a problem when testing PCIe test_p0013, log as follows:
                [   92.967294]  404 : 
                [   92.967508] PCIe Unaligned access, Norm mem   
                [   92.970259]          START  
                [   92.976420] 
                [   92.976420]  The BAR value of bdf 10000
                [   92.980157]  is 1021 
                [   92.986945] 
                [   92.986945]  BAR size is 10
                [   92.990058] Bar size less than remap requested size
                [   92.995055] : Result:  -SKIPPED- 0 
                [   93.004883]          END  
    Why to test using TYPE01_BAR, BAR0 is IO space. My hack as follows, which can pass. Please check

diff --git a/sbsa-acs-drv/files/test_pool/pcie/test_p004.c b/sbsa-acs-drv/files/test_pool/pcie/test_p004.c
index 5056c28..896c9f5 100644
--- a/sbsa-acs-drv/files/test_pool/pcie/test_p004.c
+++ b/sbsa-acs-drv/files/test_pool/pcie/test_p004.c
@@ -36,31 +36,40 @@ payload(void)
   uint32_t bdf;
   uint32_t bar_reg_value;
   uint32_t bar_value;
-  uint32_t bar_size;
+  uint32_t bar_size = 0;
   char    *baseptr;
   uint32_t index = val_pe_get_index_mpid(val_pe_get_mpid());
   uint32_t test_skip = 1;
   uint32_t test_fail = 0;
+  uint32_t offset = 0x10;

   count = val_peripheral_get_info(NUM_SATA, 0);

+    printk(KERN_EMERG "yinxuan: func=%s, line=%d, count=%d\n", __func__, __LINE__, count);
   while (count--) {
       base = val_peripheral_get_info(SATA_BASE1, count);
       bdf = val_peripheral_get_info(SATA_BDF, count);

-      val_pcie_read_cfg(bdf, TYPE01_BAR, &bar_value);
-      val_print(AVS_PRINT_DEBUG, "\n The BAR value of bdf %x", bdf);
-      val_print(AVS_PRINT_DEBUG, " is %x ", bar_value);

-      /* Determine the BAR size */
-      val_pcie_write_cfg(bdf, TYPE01_BAR, 0xFFFFFFF0);
-      val_pcie_read_cfg(bdf, TYPE01_BAR, &bar_reg_value);
-      bar_reg_value = bar_reg_value & 0xFFFFFFF0;
-      bar_size = ~bar_reg_value + 1;
-      val_print(AVS_PRINT_DEBUG, "\n BAR size is %x", bar_size);

-      /* Restore the original BAR value */
-      val_pcie_write_cfg(bdf, TYPE01_BAR, bar_value);
+        for (offset = 0x10; offset <= 0x24; offset += 4) {
+            val_pcie_read_cfg(bdf, offset, &bar_value);
+            printk(KERN_EMERG "The BAR value of bdf 0x%x is 0x%x \n", bdf, bar_value);
+
+            if (bar_value & 0x1)
+                continue;
+
+            /* Determine the BAR size */
+            val_pcie_write_cfg(bdf, offset, 0xFFFFFFF0);
+            val_pcie_read_cfg(bdf, offset, &bar_reg_value);
+
+            bar_reg_value = bar_reg_value & 0xFFFFFFF0;
+            bar_size = ~bar_reg_value + 1;
+            val_print(AVS_PRINT_DEBUG, " BAR size is %x \n", bar_size);
+
+            /* Restore the original BAR value */
+            val_pcie_write_cfg(bdf, offset, bar_value);
+        }

       /* Check if bar supports the remap size */
       if (1024 > bar_size) {

rajatgoyal47, 2021-02-01:
Hi @YinXuan-png,

Thanks for finding the issue. Fix for this issue has been merged as part of #127.
test_p004 is updated to check only if the BAR is memory address space mapped and not IO address space mapped.

Regards,
Rajat

closed 2021-02-09 by gowthamsiddarthd

YinXuan-png, 2021-02-22:
Hi Rajat, Gowtham
    The new code test_p004 can't resolve the problem I brought up before. Please review my patch.
```c
diff --git a/sbsa-acs-drv/files/test_pool/pcie/test_p004.c b/sbsa-acs-drv/files/test_pool/pcie/test_p004.c
index 5056c28..896c9f5 100644
--- a/sbsa-acs-drv/files/test_pool/pcie/test_p004.c
+++ b/sbsa-acs-drv/files/test_pool/pcie/test_p004.c
@@ -36,31 +36,40 @@ payload(void)
uint32_t bdf;
uint32_t bar_reg_value;
uint32_t bar_value;
- uint32_t bar_size;
+ uint32_t bar_size = 0;
char *baseptr;
uint32_t index = val_pe_get_index_mpid(val_pe_get_mpid());
uint32_t test_skip = 1;
uint32_t test_fail = 0;
+ uint32_t offset = 0x10;

count = val_peripheral_get_info(NUM_SATA, 0);

+ printk(KERN_EMERG "yinxuan: func=%s, line=%d, count=%d\n", func, LINE, count);
while (count--) {
base = val_peripheral_get_info(SATA_BASE1, count);
bdf = val_peripheral_get_info(SATA_BDF, count);

- val_pcie_read_cfg(bdf, TYPE01_BAR, &bar_value);
- val_print(AVS_PRINT_DEBUG, "\n The BAR value of bdf %x", bdf);
- val_print(AVS_PRINT_DEBUG, " is %x ", bar_value);

- /* Determine the BAR size */
- val_pcie_write_cfg(bdf, TYPE01_BAR, 0xFFFFFFF0);
- val_pcie_read_cfg(bdf, TYPE01_BAR, &bar_reg_value);
- bar_reg_value = bar_reg_value & 0xFFFFFFF0;
- bar_size = ~bar_reg_value + 1;
- val_print(AVS_PRINT_DEBUG, "\n BAR size is %x", bar_size);

- /* Restore the original BAR value /
- val_pcie_write_cfg(bdf, TYPE01_BAR, bar_value);
+ for (offset = 0x10; offset <= 0x24; offset += 4) {
+ val_pcie_read_cfg(bdf, offset, &bar_value);
+ printk(KERN_EMERG "The BAR value of bdf 0x%x is 0x%x \n", bdf, bar_value);
+
+ if (bar_value & 0x1)
+ continue;
+
+ / Determine the BAR size /
+ val_pcie_write_cfg(bdf, offset, 0xFFFFFFF0);
+ val_pcie_read_cfg(bdf, offset, &bar_reg_value);
+
+ bar_reg_value = bar_reg_value & 0xFFFFFFF0;
+ bar_size = ~bar_reg_value + 1;
+ val_print(AVS_PRINT_DEBUG, " BAR size is %x \n", bar_size);
+
+ / Restore the original BAR value */
+ val_pcie_write_cfg(bdf, offset, bar_value);
+ }

   /* Check if bar supports the remap size */
   if (1024 > bar_size) {
```
    I think that we should modfy bar offset, not bdf. Please check it. Thanks.
Regard,
YinXuan