ARM-software/sbsa-acs #133: Linking tests with SBSA/BSA requirements

mike-miller-microsoft-com, 2021-02-18:
I'm feeling challenged linking the tests with the SBSA/BSA requirements. There is docs/testcase-checklist.md, which is what I'm looking for, but it's references to the documentation sections seems to be out of date with the latest documentation changes.

Would it be helpful to add the rule IDs (like RPCI_IN_01) to that table - assuming those IDs are stable across versions? Or is there some other way of linking the testcases to the rules that I've missed?

hrw, 2021-03-08:
I also wonder can tests information be more granular and precise?

```
   4 : Check PE Granule Support 
       4KB: 	PASS
       64KB: 	PASS
   7 : Check Cryptographic extensions    
       AES:	PASS
       SHA1:	PASS
       SHA256:	PASS
       SHA3: 	PASS
       SHA512: 	PASS
       SM3: 	SKIP
       SM4: 	SKIP
  10 : Check AARCH64 Exception levels implementation
       EL0: 	PASS
       EL1: 	PASS
       EL2: 	PASS
       EL3: 	PASS

```

gowthamsiddarthd, 2021-03-08:
Hi @MikeMiller567,

The Current SBSA ACS tests the scenarios mentioned by the SBSA 6.0 specification and is not directly mapped to BSA v1.0 specification. The testcase-checklist points to the section in the document as SBSA 6.0 spec are not rule-based. So we don’t have a direct 1:1 mapping of the SBSA section and rules. 

Please map the testcase-checklist with SBSA 6.0 spec and not 6.1 spec. We are working on BSA ACS separately and the tests would be mapped based on the BSA rules. 

Regards,
Gowtham

MikeMiller567, 2021-03-08:
Thanks Gowtham, I'll take that approach for now.

hrw, 2021-03-10:
I wonder will something like this be usable:

```
      ***  Starting PE tests ***  
  1          : Check for number of PE            : Result:  PASS 
  2   B_PE_03: Check for SIMD extensions         : Result:  PASS 
  3 S_L3PE_02: Check for 16-bit ASID support     : Result:  PASS 
  4          : Check PE Granule Support 
      B_PE_04:     4K                            : Result:  PASS 
                  16K                            : Result:  SKIPPED 
    S_L3PE_01:    64K                            : Result:  PASS 
  5          : Check Cache Architecture          : Result:  PASS 
  6          : Check HW Coherence support        : Result:  PASS 
  7          : Check Cryptographic extensions
      B_PE_06     AES                            : Result:  PASS 
      B_PE_06     SHA1                           : Result:  PASS 
      B_PE_06     SHA256                         : Result:  PASS 
    S_L5PE_05     SHA3                           : Result:  SKIPPED
    S_L5PE_05     SHA512                         : Result:  SKIPPED
    S_L5PE_05     SM3                            : Result:  SKIPPED 
    S_L5PE_05     SM4                            : Result:  SKIPPED 
  8   B_PE_07: Check Little Endian support       : Result:  PASS 
 10          : Check Exception levels
      B_PE_08     EL0                            : Result:  PASS 
      B_PE_08     EL1                            : Result:  PASS 
    S_L3PE_03     EL2                            : Result:  PASS 
    S_L3PE_03     EL3                            : Result:  PASS 
 11   B_PE_10: Check PMU Overflow signal         : Result:  PASS 
 12          : Check number of PMU counters      : Result:  PASS 
 13          : Check Synchronous Watchpoints     : Result:  PASS 
 14          : Check number of Breakpoints       : Result:  PASS 
 15          : Check Arch symmetry across PE     : Result:  PASS 
 17          : Check CRC32 instruction support   : Result:  PASS 
 18          : Check SPE if implemented          : Result:  SKIPPED
 19 S_L4PE_01: Check for RAS extension           : Result:  FAIL
 20 S_L4PE_02: Check for 16-Bit VMID             : Result:  FAIL
 21 S_L4PE_03: Check for Virtual host extensions : Result:  FAIL
 22          : Check for pointer signing         : Result:  FAIL
 23 S_L5PE_07: Check for nested virtualization   : Result:  FAIL
 24          : Support Page table map size change: Result:  FAIL
 25 S_L5PE_06: Stage 2 control of mem and cache  : Result:  FAIL
 26 S_L5PE_04: Check Activity monitors extension : Result:  FAIL
 28          : Check AccessFlag DirtyState Update: Result:  FAIL
 29          : Check Speculation Restriction     : Result:  FAIL
 30          : Check Speculative Str Bypass Safe : Result:  FAIL
 31          : Check PEs Implement SB Barrier    : Result:  FAIL
 32          : Check PE Impl CFP,DVP,CPP RCTX    : Result:  FAIL
 33          : Check Branch Target Support       : Result:  FAIL
 34          : Check Protect Against Timing Fault: Result:  FAIL
 35 S_L6PE_06: Check Enhanced Virtualization Trap: Result:  FAIL
 36          : Check PMU Version Support         : Result:  FAIL
```

Note that is was done by hand so may have wrong ids in some places. It is more to show possible output.

gowthamsiddarthd, 2021-03-11:
Hi @hrw,

Thank you for taking the time and providing the possible output. We have planned to adopt a similar style having test number, rule ID, test description, and status of the test.

Regards,
Gowtham

closed 2021-03-12 by gowthamsiddarthd