ARM-software/sbsa-acs #134: Questions on PCIE test_p006

YinXuan-png, 2021-02-19:
### Hi,
### I had met a problem when testing PCIe test_p006. In payload function,the dma_addr from the function val_dma_device_get_dma_addr is not in the region of base from pal_smmu_check_device_iova, so the test fail. But I modify something of the file named sbsa-dma-iommu.c, as follows:
```c
//linux/include/linux/sbsa-iommu.h
#define SBSA_IOVA_DMA_ARRAY_LEN     64

//linux/drivers/iommu/sbsa-dma-iommu.c
int
sbsa_is_domain_monitored(struct iommu_domain *dom)
{
    return 1;
}
```
### I know that sbsa_is_domain_monitored is for SATA. Becase in my environmen, only one SATA device exits. I think sbsa_is_domain_monitored should return 1. I added some debug logs.

```c
void
sbsa_iommu_iova_save_addr(dma_addr_t addr, unsigned int length)
{

	static unsigned int index = 0;
	g_sbsa_iova_array[index].dma_addr = addr;
	g_sbsa_iova_array[index].dma_len = length;

        printk(KERN_EMERG "func=%s, line=%d, index=0x%x, add=0x%llx, length=0x%x \n", __func__, __LINE__, index, addr, length);
	index++;
        dump_stack();
	if (index >= SBSA_IOVA_DMA_ARRAY_LEN)
		index = 0;
}
```
### Logs show that dma_addr is belong to SATA device.  I don't know  why. Please give us som help.
### Thanks.

YinXuan-png, 2021-02-24:
Hi
    My new patch as follows:
```c
diff --git a/drivers/iommu/sbsa-dma-iommu.c b/drivers/iommu/sbsa-dma-iommu.c
index 19f0249d3397..c72cba3e52ab 100644
--- a/drivers/iommu/sbsa-dma-iommu.c
+++ b/drivers/iommu/sbsa-dma-iommu.c
@@ -31,14 +31,13 @@
 #include <linux/vmalloc.h>
 #include <linux/property.h>
 #include <linux/libata.h>
-
+#include <scsi/scsi_device.h>
+#include <scsi/scsi_host.h>
+#include <scsi/scsi_transport.h>
 #include <linux/sbsa-iommu.h>

-
 sbsa_iova_array    g_sbsa_iova_array[SBSA_IOVA_DMA_ARRAY_LEN];
-
-struct iommu_domain *g_sbsa_iommu_domain;
-
+struct iommu_domain *g_sbsa_iommu_domain = NULL;
 struct iova_domain *cookie_iovad_sbsa(struct iommu_domain *domain);

 /**
@@ -88,6 +87,38 @@ sbsa_iommu_dma_get_iova(struct device *dev, unsigned long long *base, unsigned l
 }
 EXPORT_SYMBOL(sbsa_iommu_dma_get_iova);

+
+static int sbsa_get_sata_dev(void)
+{
+    int ret = -1;
+   struct Scsi_Host   *shost;
+   struct ata_port    *ap;
+   struct scsi_device *sdev = NULL;
+   unsigned int i = 0;
+    struct device *dev;
+
+   do {
+       shost = scsi_host_lookup(i++);
+       if (shost) {
+           sdev = NULL;
+           ap = ata_shost_to_port(shost);
+           if ((ap == NULL) || (ap->dev == NULL))
+               continue; //Not a ATA port
+           do {
+               /* get the device connected to this host */
+               sdev = __scsi_iterate_devices(shost, sdev);
+               if (sdev) {
+                    g_sbsa_iommu_domain = iommu_get_domain_for_dev(ap->dev);
+                    ret = 0;
+               }
+           } while(sdev);
+           scsi_host_put(shost);
+       }
+   } while(shost);
+
+    return ret;
+}
+
 /**
   @brief   This API is used to indicate which device IOMMU transactions are to be monitored.
            1. Caller       -  Platform Abstraction Layer.
@@ -114,6 +145,11 @@ EXPORT_SYMBOL(sbsa_iommu_dev_stop_monitor);
 int
 sbsa_is_domain_monitored(struct iommu_domain *dom)
 {
+    int ret = 0;
+
+    if (g_sbsa_iommu_domain == NULL )
+        ret = sbsa_get_sata_dev();
+
    if (g_sbsa_iommu_domain == dom)
        return 1;

@@ -191,8 +227,7 @@ EXPORT_SYMBOL(sbsa_scsi_sata_get_dma_addr);
 void
 sbsa_scsi_sata_fill_dma_addr(struct scatterlist *sg)
 {
-
-        sbsa_dma_addr = cpu_to_le64(sg_dma_address(sg));
-        sbsa_dma_len  = cpu_to_le32(sg_dma_len(sg));
+    sbsa_dma_addr = cpu_to_le64(sg_dma_address(sg));
+    sbsa_dma_len  = cpu_to_le32(sg_dma_len(sg));
 }
-EXPORT_SYMBOL(sbsa_scsi_sata_fill_dma_addr);
\ No newline at end of file
+EXPORT_SYMBOL(sbsa_scsi_sata_fill_dma_addr);

```

chetan-rathore, 2022-06-22:
Thanks @YinXuan-png for the patch and apologies for the late reply. 

We are checking this test on the recent SR ACS release and will share updates soon.

Thanks,
ACS

chetan-rathore, 2022-10-14:
Hi @YinXuan-png, 

The changes proposed in the patches are integrated into linux-acs (https://gitlab.arm.com/linux-arm/linux-acs/-/blob/master/kernel/src/0001-BSA-ACS-Linux-5.15.patch) 

Can you build the latest SR image (https://github.com/ARM-software/arm-systemready/tree/main/SR#steps-to-build-systemready-sr-acs-live-image-1)  and verify the test at your end?

Thanks,
ACS Team

chetan-rathore, 2022-11-04:
Hi @YinXuan-png,

The latest pre-built image for SR which is based on Linux kernel 6.0 is available: https://github.com/ARM-software/arm-systemready/tree/main/SR/prebuilt_images/v22.10_1.1.0

Please reopen this ticket if any issue is observed with the image.

Thanks,
ACS team

closed 2022-11-04 by chetan-rathore