ARM-software/sbsa-acs #135: Error reporting may need some polish

hrw, 2021-02-20:
At this moment everything printed during run on Sbsa.efi is mixed with each other. Without any sane control over it:

```
 SMMU_INFO: Number of SMMU CTRL       :    1 

      val_smmu_init: only SMMUv3.x supported, skipping smmu 0    
      smmu_set_state: smmu unsupported      Peripheral: Num of USB controllers   :    2 
 Peripheral: Num of SATA controllers  :    4 
```

While it would be more readable in such form:

```
 SMMU_INFO: Number of SMMU CTRL       :    1
      val_smmu_init: only SMMUv3.x supported, skipping smmu 0
      smmu_set_state: smmu unsupported
 Peripheral: Num of USB controllers   :    2 
 Peripheral: Num of SATA controllers  :    4 

```

So normal output is printed as it should while error messages are printed between them.

hrw, 2021-02-20:
Also I think that all those indents done by adding spaces should be connected to val_print() level and handled there. Would be more unified.

hrw, 2021-02-20:
On higher log levels "\n" should be printed after test name to have space for all debug information

hrw, 2021-02-20:
This is how it looks now:

[hc-l6.log](https://github.com/ARM-software/sbsa-acs/files/6016654/hc-l6.log)

Those START/END messages are something I would remove.

hrw, 2021-02-20:
Notes:
- nearly all output lines starting with "\n" should have it at the
- START/END messages should be removed
- starting spaces/tabs should be removed and be added inside of val_print() function
- maybe instead of extra spaces add DBG/WARN/ERR/INFO/etc prefix
- reverse meaning of verbosity option ("-v 1" == be quiet, "-v 5" == print all)
- some information texts should not repeat (like "GIC version from ACPI table = 3" in GIC tests)

edhay, 2021-02-21:
Hi Marcin,

Thank you for the improvement comments.
We will review the entire code on the same lines and polish the logs further.

ACS Support Team

hrw, 2021-03-08:
I wonder is someone using "-l 5" or "-l 4" arguments.

Level 5 only is useless:

```
Shell> fs0:Sbsa.efi -l 3 -v 5

 SBSA Architecture Compliance Suite 
    Version 3.0  

 Starting tests for level  3 (Print level is  5)

 Creating Platform Information Tables 

      ***  Starting PE tests ***  
   1 :    2 :    3 :    4 :    5 :    6 :    7 :    8 :   10 :   11 :   12 :   1
3 :   14 :   15 :   16 :   17 : 
      ***  Starting GIC tests ***  
 101 :  102 : 
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 
 103 :  104 : 
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 

      One or more GIC tests failed. Check Log 

      *** Starting Timer tests ***  
 201 :  202 :  203 :  204 :  205 :  206 : 
       PE Timers are not always-on.       
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 
 207 : 
      *** Starting Watchdog tests ***  
 301 :  302 :  303 : 
      *** Starting PCIe tests ***  
 401 :  402 :  403 : 
      *** Starting Power and Wakeup semantic tests ***  
 501 :  502 :  503 :  504 :  505 : 
      *** Starting Peripheral tests ***  
 601 :  602 : 
       Read PCIe_CFG: ECAM Base is zero  603 : - 604 : 
       Test Message                       605 : 
      Memory access check fails at address = 0x104C0000 
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 

    One or more Peripheral tests have failed...

      *** Starting IO Virtualization tests ***  

      *** Starting PCIe Exerciser tests ***  

      *** SBSA tests complete. Reset the system. *** 

Shell> 
```

Level 4 is similarly useless:
```
Shell> fs0:Sbsa.efi -l 3 -v 4

 SBSA Architecture Compliance Suite 
    Version 3.0  

 Starting tests for level  3 (Print level is  4)

 Creating Platform Information Tables 

      ***  Starting PE tests ***  
   1 :    2 :    3 :    4 :    5 :    6 :    7 :    8 :   10 :   11 :   12 :   1
3 :   14 :   15 :   16 :   17 : 
      ***  Starting GIC tests ***  
 101 :  102 : 
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 
 103 :  104 : 
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 

      One or more GIC tests failed. Check Log 

      *** Starting Timer tests ***  
 201 :  202 :  203 :  204 :  205 : 
       v8.1 VHE not supported on this PE : Result:  -SKIPPED- 1 
 206 : 
       PE Timers are not always-on.       
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 
 207 : 
       No System timers are defined      : Result:  -SKIPPED- 1 

     *** Skipping remaining System timer tests *** 

      *** Starting Watchdog tests ***  
 301 :  302 :  303 : : Result:  -SKIPPED- 1 

      *** Starting PCIe tests ***  
 401 :  402 :  403 : 
      *** Starting Power and Wakeup semantic tests ***  
 501 :  502 :  503 :  504 :  505 : 
       No system timers implemented      : Result:  -SKIPPED- 1 

      *** Starting Peripheral tests ***  
 601 :  602 : 
       Read PCIe_CFG: ECAM Base is zero 
       WARN: SATA CTRL ECAM access failed 0  
       Re-checking SATA CTRL using PciIo protocol        603 :   604 : 
       Test Message                       605 : 
      Memory access check fails at address = 0x104C0000 
       Failed on PE -    0 for Level=  3 : Result:  --FAIL-- 1 

    One or more Peripheral tests have failed...

      *** Starting IO Virtualization tests ***  

     No SMMU Controller Found, Skipping SMMU tests...

      *** Starting PCIe Exerciser tests ***  
Exerciser Sbsa compliance is only from Level 4 

      *** SBSA tests complete. Reset the system. *** 

Shell> 
```

Maybe they both should be dropped in their current form?

Then output levels be done in such order:

1. AVS_PRINT_TEST (default)
2. AVS_PRINT_WARN
3. AVS_PRINT_ERR
4. AVS_PRINT_DEBUG
5. AVS_PRINT_INFO

hrw, 2021-03-08:
And those constants are defined in 3 different files...

hrw, 2021-09-08:
#146 does it better.

closed 2021-09-08 by hrw