ARM-software/sbsa-acs #139: Question on SBSA ACS exerciser test e015

Jingshan-cwm, 2021-03-01:
The test seems to test config/bar space to check ordering/gather rules on each PCIE EP.
Test will read/write 1B/2B/4B/(8B if needed) predefined sequence on EP, by accessing continuous address and constant address.

In our test environment (not exerciser test, cause we have not enabled sbsa exerciser test yet),
It is pass to test the pattern on aligned access 1B/2B/4B/8B.
But when we do 2B un-alignment write, we got alignment error exception.
We set the memory as device memory with DEVICE_nGnRnE attribute, and 64bit mmio address is prefechable.

R ESR_EL3  00000000 96000061
R FAR_EL3  00000200 1ff00001

In my understanding unaligned access in the test is something as below, May I know the un-alignment expectations? Does I lose something

Per ARM reference document, unaligned access wit device attribute will cause exception,
So i think, the test with unaligned access 

1. Will not test odering and gathering rule.
2. No exception handler, test will stop

Jingshan-cwm, 2021-03-02:
Per arm spec  DDI0487G_a_armv8_arm.pdf 
https://developer.arm.com/documentation/ddi0487/ga , Chapter B2.7.2 Device memory, it is not allowed to access device memory with address unaligned:

***All accesses to memory with any Device memory attribute must be aligned. Any unaligned access generates
an Alignment fault at the first stage of translation that defined the location as being Device.***

But in this test, PCIe configure/mmio space is configured as device memory 
https://github.com/ARM-software/sbsa-acs/blob/master/test_pool/exerciser/test_e015.c
```c
    /* Map mmio space to ARM device memory in MMU page tables */
    baseptr = (char *)val_memory_ioremap((void *)e_data.bar_space.base_addr, 512, DEVICE_nGnRnE);
```
And the tests issue unaligned device memory access with 16B/32B data function, <<link to the code>> 

```c
     /* Test Scenario 1 : Transactions on aligned address */
    fail_cnt += test_sequence_1B((uint8_t *)baseptr, 1, instance);
    fail_cnt += test_sequence_2B((uint16_t *)baseptr, 1, instance);
    fail_cnt += test_sequence_4B((uint32_t *)baseptr, 1, instance);
    fail_cnt += test_sequence_8B((uint64_t *)baseptr, 1, instance);

    /* Test Scenario 2 : Transactions on PCIe misaligned address */
    baseptr += 1;
    fail_cnt += test_sequence_2B((uint16_t *)baseptr, 1, instance);
    fail_cnt += test_sequence_4B((uint32_t *)baseptr, 1, instance);
    fail_cnt += test_sequence_8B((uint64_t *)baseptr, 1, instance);

    /* Test Scenario 3 : Transactions on same address */
    fail_cnt += test_sequence_1B((uint8_t *)baseptr, 0, instance);
    fail_cnt += test_sequence_2B((uint16_t *)baseptr, 0, instance);
    fail_cnt += test_sequence_4B((uint32_t *)baseptr, 0, instance);
    fail_cnt += test_sequence_8B((uint64_t *)baseptr, 0, instance);

    /* Test Scenario 4 : Transactions on same misaligned PCIe address */
    baseptr += 3;
    fail_cnt += test_sequence_2B((uint16_t *)baseptr, 0, instance);
    fail_cnt += test_sequence_4B((uint32_t *)baseptr, 0, instance);
    fail_cnt += test_sequence_8B((uint64_t *)baseptr, 0, instance);
```
The test are sure to fail with exception.

Jingshan-cwm, 2021-03-08:
Any comments？

chetan-rathore, 2021-03-09:
Hi @Jingshan-cwm  

The test case intends to verify the transactions order and gathering rule for the PCIe. We agree when PCIe space is mapped to Device memory the unaligned access check will result in exception. We will modify the test and merge the changes soon.

Thanks,
Chetan

gowthamsiddarthd, 2021-03-11:
Hi @Jingshan-cwm,

The changes have been published as part of the PR: https://github.com/ARM-software/sbsa-acs/pull/150

Regards,
Gowtham

closed 2021-03-17 by gowthamsiddarthd