ARM-software/sbsa-acs #140: "303 : Check NS Watchdog Revision" is not consistent

hrw, 2021-03-03:
QEMU SBSA reference platform is not yet even level 3 compliant but we run tests for levels 3-6 to check how situation looks.

"303: Check NS Watchdog Revision" test skips on levels 3-4 and passes on level 5-6. IMHO it should behave same way on each level as neither hardware nor firmware change between levels.

Logs: https://snapshots.linaro.org/ldcg/sbsa-acs/14/

hrw, 2021-03-03:
ah. test_w003.c has it on skip for lower levels:

```
  if (g_sbsa_level < 5) {
      val_set_status(index, RESULT_SKIP(g_sbsa_level, TEST_NUM, 01));
      return;
  }      ```

That feels weird. 

Maybe there should be a way to simply not run tests which are not valid for requested level?

hrw, 2021-03-03:
With something like this:
```
09:59 (167s) hrw@puchatek:test_pool$ git diff
diff --git a/test_pool/timer_wd/test_w003.c b/test_pool/timer_wd/test_w003.c
index fb7f4a9..df34e9a 100644
--- a/test_pool/timer_wd/test_w003.c
+++ b/test_pool/timer_wd/test_w003.c
@@ -82,6 +82,9 @@ w003_entry(uint32_t num_pe)
 {
     uint32_t status = AVS_STATUS_FAIL;
 
+    if (g_sbsa_level < 5) {
+       return AVS_STATUS_SKIP;
+    }
     num_pe = 1;  //This test is run on single processor
 
     status = val_initialize_test(TEST_NUM, TEST_DESC, num_pe, g_sbsa_level);
```

hrw, 2021-03-03:
Or even:

```
10:19 (5s) hrw@puchatek:sbsa-acs$ git diff val/
diff --git a/val/src/avs_wd.c b/val/src/avs_wd.c
index f9e28be..0761051 100644
--- a/val/src/avs_wd.c
+++ b/val/src/avs_wd.c
@@ -46,7 +46,10 @@ val_wd_execute_tests(uint32_t level, uint32_t num_pe)
 
   status = w001_entry(num_pe);
   status |= w002_entry(num_pe);
-  status |= w003_entry(num_pe);
+
+  if (level > 4) {
+      status |= w003_entry(num_pe);
+  }
 
   if (status != 0)
     val_print(AVS_PRINT_TEST, "\n      ***One or more tests have failed... *** \n", 0);
10:19 (2s) hrw@puchatek:sbsa-acs$ 
```

chetan-rathore, 2021-08-05:
The issue is a fixed with #142. Closing the issue

closed 2021-08-05 by chetan-rathore