ARM-software/sbsa-acs #144: Amount of PMU counters

hrw, 2021-03-03:
BSA v1.0 in B_PE_09 says:

> PEs must implement the FEAT_PMUv3p1 extension, and the base system must expose a minimum of four
programmable PMU counters to the operating system.

SBSA v6.0 said:

>  Each PE must implement a minimum of six programmable PMU counters.

While SBSA v6.1 in S_L3_01 says:

> The rules from Arm BSA, [5] as listed in Section 1.7.1 must be implemented.

Which means B_PE_09 must be implemented so just four PMU counters.

Test PE c012 checks for six.

Which value is proper then?

prasanth-pulla, 2021-03-03:
Marcin - BSA has 4 for OS and 2 for Hypervisor... And there is a note later in BSA which says this - 

"The combined requirements of operating system and hypervisor views is the implementation of a minimum
of six programmable PMU counters."

SBSA asks for OS and Hypervisor views to be implemented... So, the number remains 6 for SBSA 6.1 too...

hrw, 2021-03-03:
OK, we have to add two more PMU counters to QEMU SBSA ref then.

hrw, 2021-03-03:
For reference:

B_PE_21 adds those two for hypervisor.

closed 2021-03-03 by hrw