ARM-software/sbsa-acs #149: test 603 output on serial console is weird

hrw, 2021-03-08:
Output of test 603 looks weird on serial console. Tested both on QEMU SBSA reference platform and on Solidrun HoneyComb:

```
 603 : Check SBSA UART register offsets  A: Result:  PASS
 604 : Check GENERIC UART Interrupt
       Test Message                      : Result:  PASS
```

```
 603 : Check SBSA UART register offsets  0: Result:  PASS
 604 : Check GENERIC UART Interrupt      : Result:  PASS
```

```
: Result:  PASS A UART register offsets
 604 : Check GENERIC UART Interrupt      : Result:  PASS
```

```
 603 : Check SBSA UART register offsets  f: Result:  PASS 
 604 : Check GENERIC UART Interrupt      : Result:  PASS
```

At same time it looks ok on graphical output.

chetan-rathore, 2021-09-08:
Hi @hrw, 

The issue has been fixed with #178.

Can you please check from your side and close the issue.

Thanks,
ACS team

hrw, 2021-09-08:
Looks OK:

```
      *** Starting Peripheral tests ***  
 601 : Check USB CTRL Interface via PCIe : Result:  PASS 
 602 : Check SATA CTRL Interface via PCIe
       Read PCIe_CFG: ECAM Base is zero 
       WARN: SATA CTRL ECAM access failed 0  
       Re-checking SATA CTRL using PciIo protocol       : Result:  PASS 
 603 : Check SBSA UART register offsets  : Result:  PASS 
 604 : Check GENERIC UART Interrupt      
       Test Message                      : Result:  PASS 
 605 : Memory Access to Un-Populated addr
      Memory access check fails at address = 0x104C0000 
       Failed on PE -    0 for Level=  6 : Result:  --FAIL-- 1 

    One or more Peripheral tests have failed...```

closed 2021-09-08 by hrw