ARM-software/sbsa-acs #167: CNTP_CVAL issue

pullahla1234, 2021-06-10:
当前我们CNTP_CVAL这个寄存器实现的是原子写64bit，即需要同时写入0x20以及0x24，寄存器才会生效，但是测试套件测试要求高低32bit都要支持读写，查看spec有这样一段描述，如果实现支持64位原子访问，则CNTP_CVAL寄存器必须作为原子64位值访问。所以对现在测试套件验证32bit读写是有 疑问的。
<E955DC06@E9BBC46D.8811C060>
<8F561C7F@98275B63.8811C060>

pullahla1234, 2021-06-10:
Question 2: currectly, we implemented atomic write for CNTP_CVAL 64 bit register, the register value only valid after write 0x20 and 0x24 both, but the test suite require that the register HIGH and LOW 32 bit register could support read/write, the test suite test process is not consistent with the spec description  as below. Please help check.

prasanth-pulla, 2021-06-11:
Hi,

Question #1 : The change to do both 32-bit writes before doing read is done in above checkin.

Question #2: Even if 64-bit atomic write is implemented, Arm ARM mentions that for this memory map interface, the individual 32-bit writes must still be supported. This is necessary for Memory mapped interface.

closed 2021-06-11 by prasanth-pulla