ARM-software/sbsa-acs #175: Generating HVP testplans and coverage?

MikeMiller567, 2021-07-21:
My team is using Synopsys Verification Planner for tracking progress and I would like to use it to plan/track running each of the sbsa tests as they run/pass in my baremetal environment.

I was wondering if anyone had already done something (like a script or something) to create the .hvp plan file and extract/record coverage.

gowthamsiddarthd, 2021-08-12:
Hi @MikeMiller567,

Currently, we do not have any support to extract or measure the coverage. 
Please contact Synopsys representative for support on this.
	
Regards,
Gowtham

MikeMiller567, 2021-08-12:
Thanks for getting back to me @gowthamsiddarthd

I'll put together a python program to scrape the data out of the source code and produce the tracking data I need.

closed 2021-08-12 by MikeMiller567