ARM-software/sbsa-acs #222: The exception or system weird state caused by RootPort Memory Access tests (448 and 449)

sunnywang-arm, 2022-02-03:
Just create this issue for having a record.
For the details, please check the fix: https://github.com/ARM-software/sbsa-acs/pull/221

chetan-rathore, 2022-08-10:
Hi @sunnywang-arm, 

The fix for the issue is merged with #267.
RP which has an ethernet or VGA controller will be skipped from checking the out-of-range access to P/NP memory.

Thanks,
ACS team

sunnywang-arm, 2022-10-03:
We found the problem still occurs with the in-use Storage controller and found another exception issue caused by accessing the PCI Express Capability structure of a legacy device. Once the fixes get merged, the ACS team will add comments here for closing the issue.

sunnywang-arm, 2022-10-12:
Just FYI, the fix has also been verified on several systems.