ARM-software/sbsa-acs #228: SBSA 408: Check MSI(X) vectors uniqueness

sunnywang-arm, 2022-03-22:
Got the error message "Failed to get address of PCI device". It looks like something is wrong in g_peripheral_info_table. We may need to check why there is a peripheral (an entry in g_peripheral_info_table) found in the system by ACS that has MSI flag set but doesn’t have BDF (address of PCI device). 

```
[   20.678952]  408 : 
[   20.678953] Check MSI(X) vectors uniqueness   
[   20.681478] 
[   20.681478]        Failed to get address of PCI device
[   20.685908] 
[   20.685908]        Failed on PE -    0 
[   20.692508] for Level=  4 
[   20.697805] : Result:  --FAIL-- 1 
[   20.703888]  409 : 
[   20.703889] Check all MSI(X) vectors are LPIs 
[   20.706093] : Result:  PASS 
```

Since test 409 is also based on g_peripheral_info_table and the platform get the PASS result from test 409, I think we can sync test 408 with test 409 to just skip checking this kind of entry as a workaround. 

By the way, I also checked the kernel message and cat ./priroc/interrupts output messages. All the PCI peripherals (integrated controllers and one add-in card) on this platform are working fine with the default kernel boot command (NO “nomsi”).

chetan-rathore, 2022-06-17:
Merge request for the issue is raised in the linux-acs branch.

https://gitlab.arm.com/linux-arm/linux-acs/-/merge_requests/15

sunnywang-arm, 2022-06-17:
Closing this issue as we don't see the failure with the test image. Thanks, @chetan-rathore and Sujana M.

closed 2022-06-17 by sunnywang-arm