ARM-software/sbsa-acs #233: SBSA EFI can't save content into SbsaResults.log - Sized 0KB

sunnywang-arm, 2022-05-13:
Track this issue publicly in case others also run into the same problem. 
The issue is that the SBSA EFI test is done but SbsaResults.log in USB drive shows empty. 
Using UEFI Shell Redirection "Sbsa.efi > SbsaResults.log" works, but using "sbsa.efi -f SbsaResults.log" doesn't work.

chetan-rathore, 2022-06-22:
Hi @sunnywang-arm, 

We are already checking the issue and in process of verifying the fix. We will share the updates on root cause soon.

Thanks,
ACS

chetan-rathore, 2022-07-05:
Hi @sunnywang-arm,

The issue is observed due to smmu init code, it is required only by pcie exerciser test. As a workaround the code is moved inside exerciser test run function.

https://github.com/ARM-software/sbsa-acs/pull/251

Thanks,
ACS

chetan-rathore, 2022-07-06:
Hi @sunnywang-arm,

Changes has been merged.

Thanks,
ACS

closed 2022-07-06 by chetan-rathore