ARM-software/sbsa-acs #234: Improvement for 413 Addressability of Non-Sec masters

sunnywang-arm, 2022-05-13:
The error message is
`WARNING: The device with bdf=0xXXXXXXX doesn't support 64 bit addressing and is not behind SMMU. Please install driver for this device and test it again. If the driver is already installed, this test has failed.`

The root cause is that the ACS Linux kernel doesn't have the device driver for the device with bdf=0xXXXXXXX. 

Suggest skipping this test in automation.  People who have a need to run this test can still try to find and install the device driver and then manually run it.   

If we still need to keep this test in automation, the suggestion would be to add device drivers as many as possible. Otherwise, it is easy to run into this failure when the system has an integrated PCIe device or a PCIe card.

chetan-rathore, 2022-06-22:
Hi @sunnywang-arm,

We are checking the root cause suggested by you at our side and share the updates soon.

Thanks,
ACS

chetan-rathore, 2022-06-24:
Hi @sunnywang-arm, 

The current failure is due to the device driver not being present for some devices in SR ACS live image Linux. 
Agree with your point, that on automation image will be difficult to add system-specific drivers. We are adding code to check if driver is present for a pcie device before checking its dma attributes. In case driver is not present, skip that BDF.

Verification of code is in progress.
https://github.com/chetan-rathore/sbsa-acs/commit/f37b9df0c539db202df5b68d9b31bd12fc76f022
https://gitlab.arm.com/linux-arm/linux-acs/-/commit/38e8c5989b93bb8f9ad320633748c9ae636e7b76

chetan-rathore, 2022-06-29:
Hi @sunnywang-arm, 

Test is updated with below PR.
https://gitlab.arm.com/linux-arm/linux-acs/-/merge_requests/19
https://github.com/ARM-software/sbsa-acs/pull/250

chetan-rathore, 2022-06-30:
Hi @sunnywang-arm, 

The changes are merged.

Thanks,
ACS

sunnywang-arm, 2022-09-15:
Fix is in https://gitlab.arm.com/linux-arm/linux-acs/-/commit/6b51bb71b83769247a98c1cda3e94ca398cd61ca