ARM-software/sbsa-acs #235: System ASSERT or exception while running GIC related tests

sunnywang-arm, 2022-05-16:
Tracking this issue publicly in case others also run into the same problem.

SBSA can’t get a valid base address of the GIC distributor. val_get_gicd_base() returns 0 instead of the valid base address in ACPI MADT table, which causes system ASSERT at test case 11 (Check PMU overflow signal).  After skipping test case 11, the system will run into ASSERT at test case 103 (GIC number of security states) instead.

chetan-rathore, 2022-06-17:
Hi @sunnywang-arm, 

The issue was due to GIC INFO TABLE size was not sufficient to hold all the gic entries of MADT table of the particular system. The table size is increased as per the system requirements and the fix verification is in progress with the partner.

Thanks,
ACS team

chetan-rathore, 2022-06-21:
Hi @sunnywang-arm, 

Based on partner feedback on fix verification, PR has been raised to merge the changes