ARM-software/sbsa-acs #238: Query about PCIe test test_p043.c

tvidyasagar, 2022-05-22:
The condition in line 75 to check if a Switch or no device is present downstream works well only to find out if a switch is present but may not for the device's absence. In a system where there is a root port but no endpoint is connected to it would have PriBus=0, SecBus=1, and SubBus=1. With this, the condition in line 75 evaluates to false which is wrong. A better way to identify the device's absence would be to read DLActive status.

gowthamsiddarthd, 2022-05-24:
Hi @tvidyasagar,

Thank you for reporting this issue. This has been fixed as part of https://github.com/ARM-software/sbsa-acs/pull/239

Regards,
ACS Team

tvidyasagar, 2022-05-25:
I verified #239 and it fixes this issue.

closed 2022-05-25 by tvidyasagar