ARM-software/sbsa-acs #240: HW Coherency Tests Skipped

ashishsingha, 2022-05-27:
"Check HW Coherence support" test in test_c006 is skipped on targets that do not implement AArch32 and only implement AArch64. Because of this, despite AArch64 being implemented correctly, the test is always skipped.

chetan-rathore, 2022-06-17:
Hi @ashishsingha,

As per ArmArm. the AArch64 ID_MMFR0_EL1 reg bits are valid when AArch32 is supported by PE (section D13.2.81), hence the test is first checking for a32_support before reading the necessary bits from ID_MMFR0_EL1 register.

Thanks,
ACS team

ashishsingha, 2022-06-17:
Hi, @chetan-rathore,

It is totally valid and common for arm architecture silicon to support AArch64 only and not have AArch32 support at all. Are you saying that for this use case there cannot be any HW coherency tests?

Thanks
Ashish

chetan-rathore, 2023-01-04:
Hi @ashishsingha, 

Please find updates from our internal discussion.

The B_PE_05 rule says:
 1) All PEs are coherent and – all Armv8 PEs must be coherent within an inner shareable domain, this is mandated by the Arm ARM
 2) in the same Inner Shareable domain – the presence of all PEs, being tested by the ACS, within the same inner shareability domain is what the ACS test should assert, unfortunately, there is no architected way for this to be determined, not even via the register for cache hierarchy description.

Given the above, our opinion is that the present test should be removed, as it is not telling us much with the way it is currently implemented.

Thanks,
ACS team

chetan-rathore, 2023-02-01:
Hi @ashishsingha, 

B_PE_05 rule is covered by BSA ACS PE test 5.  
Required changes are raised in BSA ACS : https://github.com/ARM-software/bsa-acs/pull/104

Thanks,
ACS team

chetan-rathore, 2023-02-03:
The changes are merged.

closed 2023-02-03 by chetan-rathore

ashishsingha, 2023-02-03:
Thanks @chetan-rathore