ARM-software/sbsa-acs #246: BSA handles PCIe Multi-Function Device(MFD) bit

jstxtj, 2022-06-24:
Per PCIe spec, only for a multi-function device, then check remaining non-zero functions:
**Header Type Register (Offset 0Eh)**:
7 Multi-Function Device - When Set, indicates that the Device may contain multiple Functions, but not necessarily. Software is permitted to probe for Functions other than Function 0. **When Clear, software must not probe for Functions other than Function 0 unless explicitly indicated by another mechanism, such as an ARI or SR-IOV Extended Capability structure.** Except where stated otherwise, it is recommended that this bit be Set if there are multiple Functions, and Clear if there is only one Function.

It is more reasonable way to check if a specific device on a specific bus is present and if it is multi-function or not, if it's a multi-function device, then check remaining functions.  

If don't check bit 7 of the header type and scan all functions, then some single-function devices will report details for "function 0" for every function. In fact, these non-zero function devices do not exist, subsequently some BSA test cases on these devices may fail, or otherwise be problematic.

chetan-rathore, 2022-06-27:
Hi @jstxtj, 

We are checking internally on handling the MFD bit.

Thanks,
ACS

sunnywang-arm, 2022-10-14:
Had an offline discussion with @chetan-rathore and agreed with his point, so I think we can close this issue now. What do you think? @jstxtj 

In the description of MFD (bit 7) in PCIe spec, the last sentence “ it is recommended that this bit be Set if there are multiple Functions, and Clear if there is only one Function.” means MFD should honour its hardware. If SBSA uses this optimization (MFD), the HW issue where the MDF doesn't honour its hardware will be hidden.

foxmmmm, 2022-10-24:
@sunnywang-arm, @chetan-rathore , yes, close this issue.

sunnywang-arm, 2022-10-24:
@chetan-rathore please close this issue as I already confirmed this @jstxtj (@foxmmmm).

chetan-rathore, 2022-10-29:
Thanks @sunnywang-arm and @foxmmmm for closing the discussion.

closed 2022-10-29 by chetan-rathore