ARM-software/sbsa-acs #262: SBSA PE tests do not execute successfully on aarch64 machines due to secondary stack alignment issue

r0bwoo, 2022-08-02:
Attempting to run the PE tests on an aarch64 system with more than one vCPU results in the following test failures.

```
 1 : Check for number of PE                     START  : Result:  PASS 
         END  

   2 : Check for SIMD extensions                  START         PSCI_CPU_ON: failure  

       Failed on PE -    1 for Level=  4 : Result:  --FAIL-- 121 
         END  

   3 : Check for 16-bit ASID support              START         PSCI_CPU_ON: failure  

       Failed on PE -    1 for Level=  4 : Result:  --FAIL-- 121 
         END  

   4 : Check PE Granule Support                   START         PSCI_CPU_ON: failure  

       Failed on PE -    1 for Level=  4 : Result:  --FAIL-- 121 
         END  

   5 : Check Cache Architecture                   START         PSCI_CPU_ON: failure  

       Failed on PE -    1 for Level=  4 : Result:  --FAIL-- 121 
         END  

   6 : Check HW Coherence support                 START         PSCI_CPU_ON: failure  

       Failed on PE -    1 for Level=  4 : Result:  --FAIL-- 121 
         END  
...
```

Note the "PSCI_CPU_ON: failure" message. If the file `platform/pal_uefi/src/pal_pe.c` is modified to align the stack allocation from gBS->AllocatePool to 16 bytes then the PE tests are able to execute without issue on multi vCPU systems. Additionally, the arm handbook states that aarch64 must be at least 16 byte aligned where aarch32 may be at least 4 byte aligned. (https://community.arm.com/arm-community-blogs/b/architectures-and-processors-blog/posts/using-the-stack-in-aarch32-and-aarch64)

r0bwoo, 2022-08-02:
I wanted to create an issue here first before a PR to confirm if this is an actual issue.

samerhaj, 2022-08-03:
Looks like this may be the issue [PalAllocateSecondaryStack() ](https://github.com/ARM-software/sbsa-acs/blob/master/platform/pal_uefi/src/pal_pe.c#L102).

The UEFI spec says stack allocations need to be 16-byte aligned:

> 2.3.6.4 Detailed Calling Convention
> The base calling convention for the AArch64 binding is defined in the document Procedure Call Standard for the ARM
> 64-bit Architecture Version A-0.06 (or later):
> ...
> • The stack will maintain 16 byte alignment.

The [Arm 64-bit calling convention](https://github.com/ARM-software/abi-aa/releases/download/2022Q1/aapcs64.pdf) also requires this:

> 6.2.2.1 Universal stack constraints
> ...
> Additionally, at any point at which memory is accessed via SP, the hardware requires that 
>   • SP mod 16 = 0. The stack must be quad-word aligned
>  

EDK2 UEFI code does this for the DXE stack allocation [MdeModulePkg/Core/DxeIplPeim.Arm/DxeLoadFunc.c](https://github.com/tianocore/edk2/blob/master/MdeModulePkg/Core/DxeIplPeim/Arm/DxeLoadFunc.c#L52) :
```
  TopOfStack = (VOID *)((UINTN)BaseOfStack + EFI_SIZE_TO_PAGES (STACK_SIZE) * EFI_PAGE_SIZE - CPU_STACK_ALIGNMENT);
  TopOfStack = ALIGN_POINTER (TopOfStack, CPU_STACK_ALIGNMENT);
```
And [AArch64/ProcessorBind.h](https://github.com/tianocore/edk2/blob/master/BaseTools/Source/C/Include/AArch64/ProcessorBind.h#L90) :

`#define CPU_STACK_ALIGNMENT  16
`

chetan-rathore, 2022-08-05:
Hi @r0bwoo, 

Agree that the stack pointer allocated needs to be 16-byte aligned. 

The [PalAllocateSecondaryStack](https://github.com/ARM-software/sbsa-acs/blob/cca1e7729762470f7a740b0ce4d8c7b832757308/platform/pal_uefi/src/pal_pe.c#L90) function will be modified to use the  [pal_aligned_alloc](https://github.com/ARM-software/sbsa-acs/blob/cca1e7729762470f7a740b0ce4d8c7b832757308/platform/pal_uefi/src/pal_misc.c#L621) for memory allocation.

Thanks,
ACS

r0bwoo, 2022-08-10:
Just wanted to add that we have https://github.com/ARM-software/sbsa-acs/pull/265 for review. This allocates and aligns the memory to 16 bytes and also stores the original returned pointer from edk2 so it may be freed in the future if desired.

chetan-rathore, 2022-08-11:
Hi @r0bwoo,

#265 has been merged.

Thanks,
ACS team

closed 2022-08-17 by r0bwoo