ARM-software/sbsa-acs #273: sbsa-acs-master\val\src\avs_pe.c : In  "val_pe_reg_read "undefined reference to few functions (ArmReadMpidr)

sekharg1995, 2022-09-05:
While I am trying to compile pcie related test cases in baremetal environment , i am getting error as "undefined reference to  these functions related to arm 'ArmReadMpidr' , 'ArmCallSmc' " etc .
I could not find any definitions for the above mentioned functions .
So how can we resolve these . 

Thanks in advance

gowthamsiddarthd, 2022-09-07:
Hi @sekharg1995,

Can you provide more details on the below aspects
1. Complete compile/build log
2. Which branch is being used (master/baremetal_reference/release)
3. Steps used for compilation

We would require more info to get to the root cause. These are registers that are mentioned in *.S files
The failure can be due to
1. Improper makefile compilation
2. Compiler used and its version.

Regards,
Gowtham Siddarth,
The ACS team

sekharg1995, 2022-09-16:
////////////////////////////////// steps for compilation //////////////////////////////////////////////////// 
we have downloaded the complete master repo to our local work area, we have taken the test_p001.c as reference and started to compile the code in our environment.
We have included all the required header files , api.c files [all the necessary files], then we are facing these errors.

//////////////////////////////////// query 1 ///////////////////////////////////

I have not found the function definitions (ArmCallSmc , ArmReadMpidr ,ArmReadHcr etc) which are called in val_pe_get_mpid()  

for your reference ,this function - val_pe_get_mpid() is called from payload(void) in test_p001.c file 

/**
  @brief   This API provides a 'C' interface to call System register reads
           1. Caller       -  Test Suite
           2. Prerequisite -  None
  @param   reg_id  - the system register index for which data is returned
  @return  the value read from the system register.
**/
uint64_t
val_pe_reg_read(uint32_t reg_id)
{

  switch(reg_id) {
      case MPIDR_EL1:
          return ArmReadMpidr();
      case ID_AA64PFR0_EL1:
          return ArmReadIdPfr0();
      case ID_AA64PFR1_EL1:
          return ArmReadIdPfr1();
      case ID_AA64MMFR0_EL1:
          return AA64ReadMmfr0();
      case ID_AA64MMFR1_EL1:
          return AA64ReadMmfr1();
      case ID_AA64MMFR2_EL1:
          return AA64ReadMmfr2();
      case CTR_EL0:
          return AA64ReadCtr();
      case ID_AA64ISAR0_EL1:
          return AA64ReadIsar0();
      case ID_AA64ISAR1_EL1:
          return AA64ReadIsar1();
      case SCTLR_EL3:
          return AA64ReadSctlr3();
      case SCTLR_EL2:
          return AA64ReadSctlr2();
      case SCTLR_EL1:
          return AA64ReadSctlr1();
      case PMCR_EL0:
          return AA64ReadPmcr();
      case ID_AA64DFR0_EL1:
          return AA64ReadIdDfr0();
      case ID_AA64DFR1_EL1:
          return AA64ReadIdDfr1();
      case CurrentEL:
          return AA64ReadCurrentEL();
      case MDCR_EL2:
          return AA64ReadMdcr2();
      case VBAR_EL2:
          return AA64ReadVbar2();
      case CCSIDR_EL1:
          return AA64ReadCcsidr();
      case CSSELR_EL1:
          return AA64ReadCsselr();
      case CLIDR_EL1:
          return AA64ReadClidr();
      case ID_DFR0_EL1:
          return ArmReadDfr0();
      case ID_ISAR0_EL1:
          return ArmReadIsar0();
      case ID_ISAR1_EL1:
          return ArmReadIsar1();
      case ID_ISAR2_EL1:
          return ArmReadIsar2();
      case ID_ISAR3_EL1:
          return ArmReadIsar3();
      case ID_ISAR4_EL1:
          return ArmReadIsar4();
      case ID_ISAR5_EL1:
          return ArmReadIsar5();
      case ID_MMFR0_EL1:
          return ArmReadMmfr0();
      case ID_MMFR1_EL1:
          return ArmReadMmfr1();
      case ID_MMFR2_EL1:
          return ArmReadMmfr2();
      case ID_MMFR3_EL1:
          return ArmReadMmfr3();
      case ID_MMFR4_EL1:
          return ArmReadMmfr4();
      case ID_PFR0_EL1:
          return ArmReadPfr0();
      case ID_PFR1_EL1:
          return ArmReadPfr1();
      case MIDR_EL1:
          return ArmReadMidr();
      case MVFR0_EL1:
          return ArmReadMvfr0();
      case MVFR1_EL1:
          return ArmReadMvfr1();
      case MVFR2_EL1:
          return ArmReadMvfr2();
      case PMCEID0_EL0:
          return AA64ReadPmceid0();
      case PMCEID1_EL0:
          return AA64ReadPmceid1();
      case VMPIDR_EL2:
          return AA64ReadVmpidr();
      case VPIDR_EL2:
          return AA64ReadVpidr();
      case PMBIDR_EL1:
          return AA64ReadPmbidr();
      case PMSIDR_EL1:
          return AA64ReadPmsidr();
      case LORID_EL1:
          return AA64ReadLorid();
      case ERRIDR_EL1:
          return AA64ReadErridr();
      case ERR0FR_EL1:
          return AA64ReadErr0fr();
      case ERR1FR_EL1:
          return AA64ReadErr1fr();
      case ERR2FR_EL1:
          return AA64ReadErr2fr();
      case ERR3FR_EL1:
          return AA64ReadErr3fr();
      case ESR_EL2:
          return AA64ReadEsr2();
      case FAR_EL2:
          return AA64ReadFar2();
      case RDVL:
          return ArmRdvl();
      case MAIR_ELx:
          if (AA64ReadCurrentEL() == AARCH64_EL1)
            return AA64ReadMair1();
          if (AA64ReadCurrentEL() == AARCH64_EL2)
            return AA64ReadMair2();
      case TCR_ELx:
          if (AA64ReadCurrentEL() == AARCH64_EL1)
            return AA64ReadTcr1();
          if (AA64ReadCurrentEL() == AARCH64_EL2)
            return AA64ReadTcr2();
      default:
           val_report_status(val_pe_get_index_mpid(val_pe_get_mpid()), RESULT_FAIL(g_sbsa_level, 0, 0x78));
  }

  return 0x0;
}

All the functions are used , but i cannot find the definitions of the same. please help us on the same.

///////////////////////////////////////////// query 2 /////////////////////////////////////////////////

In Armsmc.S , could you please provide the details what exactly this "GCC_ASM_EXPORT(ArmCallSmc)" is doing.

[compile log.txt](https://github.com/ARM-software/sbsa-acs/files/9581102/compile.log.txt)

gowthamsiddarthd, 2022-10-04:
Hi @sekharg1995,

Looks like the functions defined in the Assembly files (*.S) files are not available to the linker to resolve the references.
Can you please check if all these .S files are included in the makefile compilation?

The reference to create Makefiles can be found in
[Makefile](https://github.com/ARM-software/sbsa-acs/blob/master/Makefile)
[val.mk](https://github.com/ARM-software/sbsa-acs/blob/master/val/val.mk)
[pal_baremetal.mk](https://github.com/ARM-software/sbsa-acs/blob/master/platform/pal_baremetal.mk)

GCC_ASM_EXPORT() & GCC_ASM_IMPORT() macros encapsulate the declarations required by ARM GCC. They are used in ARM GCC assembly files that can be used by the linker to resolve symbol references in separate object and library files.

Regards,
Gowtham

sekharg1995, 2022-10-20:
Hi Team,

Query 1:
While we are using the assembly files in our environment, the **#defines** are getting commented as per the below image 
[image]

But in the code provided by you, the #defines (macros) are not commented [ as per the below snip]
[image]

Query 2:
I am trying to build the environment for windows environment as per the steps mentioned in the snip.
In the below snip, could you please give me details about the 3rd step. What are the packages to be downloaded and how should we set the environment.
[image]

gowthamsiddarthd, 2022-11-04:
Hi @sekharg1995,

For Query #1
The code doesn't seems to be commented. The problem might be in the toolchain used.
Can you provide details on what toolchain is being used? Can you try with "aarch64-none-linux-gnu"
Details on setting up the toolchain can be found at https://github.com/ARM-software/sbsa-acs#linux-build-environment

For Query #2
Details on Package can be found at https://github.com/ARM-software/sbsa-acs#prerequisites
To reiterate, 
1. git clone the EDK2 port of libc from https://github.com/tianocore/edk2-libc
   SHA: 61687168fe02ac4d933a36c9145fdd242ac424d1
2. git submodule update --init --recursive

closed 2022-11-24 by gowthamsiddarthd