ARM-software/sbsa-acs #289: Error: unknown or missing system register name at operand 2 -- `mrs x0,mpam1_el1'

hrw, 2023-01-24:
While building sbsa-acs under Debian 'bullseye' I got this error:

```
"aarch64-linux-gnu-gcc"  -c -x assembler -imacros AutoGen.h -mlittle-endian -march=armv8.2-a+sve+profile -o /tmp/tests/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/val/SbsaValLib/OUTPUT/src/AArch64/MpamSupport.obj -I/tmp/tests/edk2/ShellPkg/Application/sbsa-acs/val/sys_arch_src/gic/its -I/tmp/tests/edk2/ShellPkg/Application/sbsa-acs/val/sys_arch_src/gic/v2 -I/tmp/tests/edk2/ShellPkg/Application/sbsa-acs/val/sys_arch_src/gic/v3/AArch64 -I/tmp/tests/edk2/ShellPkg/Application/sbsa-acs/val/sys_arch_src/gic/v3 -I/tmp/tests/edk2/ShellPkg/Application/sbsa-acs/val/sys_arch_src/gic/AArch64 -I/tmp/tests/edk2/ShellPkg/Application/sbsa-acs/val/sys_arch_src/gic -I/tmp/tests/edk2/ShellPkg/Application/sbsa-acs/val/sys_arch_src/smmu_v3 -I/tmp/tests/edk2/ShellPkg/Application/sbsa-acs/val/src -I/tmp/tests/edk2/ShellPkg/Application/sbsa-acs/val/src/AArch64 -I/tmp/tests/edk2/ShellPkg/Application/sbsa-acs/val -I/tmp/tests/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/val/SbsaValLib/DEBUG -I/tmp/tests/edk2/MdePkg -I/tmp/tests/edk2/MdePkg/Include -I/tmp/tests/edk2/MdePkg/Test/UnitTest/Include -I/tmp/tests/edk2/MdePkg/Include/AArch64 /tmp/tests/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/val/SbsaValLib/OUTPUT/src/AArch64/MpamSupport.iiii
/tmp/tests/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/val/SbsaValLib/OUTPUT/src/AArch64/MpamSupport.iiii: Assembler messages:
/tmp/tests/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/val/SbsaValLib/OUTPUT/src/AArch64/MpamSupport.iiii:31: Error: unknown or missing system register name at operand 2 -- `mrs x0,mpam1_el1'
/tmp/tests/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/val/SbsaValLib/OUTPUT/src/AArch64/MpamSupport.iiii:35: Error: unknown or missing system register name at operand 1 -- `msr mpam1_el1,x0'
/tmp/tests/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/val/SbsaValLib/OUTPUT/src/AArch64/MpamSupport.iiii:39: Error: unknown or missing system register name at operand 2 -- `mrs x0,mpam2_el2'
/tmp/tests/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/val/SbsaValLib/OUTPUT/src/AArch64/MpamSupport.iiii:43: Error: unknown or missing system register name at operand 1 -- `msr mpam2_el2,x0'
/tmp/tests/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/val/SbsaValLib/OUTPUT/src/AArch64/MpamSupport.iiii:47: Error: unknown or missing system register name at operand 2 -- `mrs x0,mpamidr_el1'
```

sbsa-acs commit:
```
commit ba28f078a3769d2df9a31dd2443af0d61f464be8 (grafted, HEAD -> master, origin/master, origin/HEAD)
Author: Rajat Goyal <rajat.goyal@arm.com>
Date:   Mon Jan 16 21:37:59 2023 +0530

    Merge pull request #288 from rajatgoyal47/master

    Prebuilt image & Changelog updated
```

Cross compiler comes from Debian:
```
root@ba383ca0d857:/tmp/tests# aarch64-linux-gnu-gcc --version
aarch64-linux-gnu-gcc (Debian 10.2.1-6) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Last working version:

```
commit 85566c6872d4dd1ab579a30b84a63dc86b9a11d8 (HEAD -> master)
Merge: 275684b 096150f
Author: chetan singh rathore <chetanrgithub@gmail.com>
Date:   Sat Jan 7 09:34:55 2023 +0530

    Merge pull request #285 from shyamanthrh/gcc12-fix

    Fix for issue #245
```

hrw, 2023-01-24:
With your way of doing code dumps I am unable to process it further.

chetan-rathore, 2023-01-24:
Hi @hrw, 

The minimum toolchain version required will be 10.3 for compiling, as mpam related system register are present from that version.

Thanks,
ACS team

closed 2023-01-26 by hrw