ARM-software/sbsa-acs #290: MPAM (PE test 15) should fail on level 7

hrw, 2023-01-26:
When I run "sbsa.efi -v 1 -l 7" on sbsa-ref (to see how far we are) I noticed that test 15 was skipped while it should fail:

```
  15 : Check MPAM PE Requirements        
       START  
       PSCI_CPU_ON: success  
       S_MPAM_PE
       Checkpoint --  2                  : Result:  SKIPPED 
       END  
```

SBSA 7.0 (DEN0029F) says:

> RS_L7MP_01 PEs must implement the MPAM extension. See FEAT_MPAM in [2].

So (IMHO) test should fail on level 7 and skip on lower ones where MPAM support is optional.

chetan-rathore, 2023-01-27:
Hi @hrw, 

In the SBSA 7.1 specification, we have two rules related to MPAM PE requirements. 
S_MPAM_PE, which is present for L5 onwards  and 
S_L7MP_01 which is for L7 onwards. 

The PE test 15 covers the S_MPAM_PE rule.  
(https://github.com/ARM-software/sbsa-acs/blob/ba28f078a3769d2df9a31dd2443af0d61f464be8/test_pool/pe/operating_system/test_c015.c#L24)

RS_MPAM_PE Implementation of the MPAM extension (FEAT_MPAM) is OPTIONAL, however if implemented, the following
minimal requirements must be met by the implementation:
• provide a minimum of 16 physical partition IDs.
• provide virtualization support with a minimal of 8 virtual partition IDs.
• provide a minimal of 2 performance monitor groups.
• provide cache portion partitioning in the last level cache.

S_L7MP_01 is covered by the MPAM test 1.
(https://github.com/ARM-software/sbsa-acs/blob/ba28f078a3769d2df9a31dd2443af0d61f464be8/test_pool/mpam/operating_system/test_mpam001.c#L25)

Thanks,
ACS team

hrw, 2023-01-27:
ok, just feels weird.

closed 2023-01-27 by hrw