ARM-software/sbsa-acs #307: Synchronize level checks?

hrw, 2023-04-03:
When SBSA ACS is run for level 5 watchdog tests pass:

```
      *** Starting Watchdog tests ***

      All Watchdog tests passed.
```

But the only watchdog test left in SBSA ACS is 701 for S_L6WD_01. So check in SbsaAvsMain.c should check for level > 5:

```
  if (g_sbsa_level > 4)
  {
    val_print(AVS_PRINT_TEST, "\n      *** Starting Watchdog tests ***  \n", 0);
    Status |= val_wd_execute_tests(g_sbsa_level, val_pe_get_num());
  }
```

Similar is with PCIe tests. They are run for "level > 5" and then check for "level < 6" which never applies.

And there are still places with "level > 2" checks where minimal supported version is level 3.

chetan-rathore, 2023-04-10:
Hi @hrw,

Changes for the issue are merged.

Thanks,
ACS team

hrw, 2023-04-11:
One left:

```
      *** Starting PCIe Exerciser tests ***  
Exerciser Sbsa compliance is only from Level 4 
```

chetan-rathore, 2023-04-17:
Hi @hrw,

Can you please open a separate issue for exerciser tests, we are having an internal discussion on which levels to run the exerciser test and it will take some time.

Others changes are already merged, so this issue can be closed.

Thanks,
ACS team

closed 2023-04-17 by hrw