ARM-software/sbsa-acs #308: test 301 lacks S_L4SM_01 and S_L5SM_01 rules

hrw, 2023-04-03:
Test 301 checks for SMMU version.

For levels 3 it is fine with SMMUv2, for level 4 it wants SMMUv3, for level 5 (and above) it wants SMMUv3.2 or higher.

Level 3 has S_L3SM_01 rule which says nothing about SMMU version but rather about it's functionality (which I assume is supported by SMMUv2+ so that's why test just goes with version).

Level 4 has S_L4SM_01 rule, level 5 has S_L5SM_01 rule explicitly for SMMU version check. Please add those tags.

chetan-rathore, 2023-04-17:
Hi @hrw,

We are working on it and the changes are expected to be raised by next week. 

Thanks,
ACS team

hrw, 2023-05-04:
The other solution would be to add separate tests for each of tags.

PR #330 (2023-05-04): SMMU test 301 & 302 test reorder and updates
  - test 302 covers S_L5SM_01, S_L5SM_02 directly. S_L3SM_01 is removed since it is waived off for its IMPDEF implications.
  - since test 302 runs for level >=5, redundant level checks and dead code is removed.
  - corrected RULE to test mapping.

Resolves: #308
commit dc075dd5 (2023-05-04): SMMU test 301 & 302 test reorder and updates

  - test 302 covers S_L5SM_01, S_L5SM_02 directly.
    S_L3SM_01 is removed since it is waived off for its IMPDEF implications.
  - since test 302 runs for level >=5, redundant level checks and
    dead code is removed.
  - corrected RULE to test mapping.

Resolves: #308