ARM-software/sbsa-acs #327: test 861:        Received exception of type: 0

hrw, 2023-04-24:
When I run "sbsa.efi -l 7 -v 1" I get this:

```
 861 : Check RootPort P&NP Memory Access
       START
       BDF - 0x100
       BDF - 0x300
       Received exception of type: 0
       Received exception of type: 0
       Received exception of type: 0
       Received exception of type: 0
       Received exception of type: 0
```

And it loops here.

"git bisect" ended here:

```
11:27 marcin@applejack:sbsa-acs$ git bisect bad
ae6d9fc7e6ed3d8cb80e1aba4ef0f0d26104ec5c is the first bad commit
commit ae6d9fc7e6ed3d8cb80e1aba4ef0f0d26104ec5c
Author: Srikar Josyula <srikar.josyula@arm.com>
Date:   Fri Apr 14 00:46:39 2023 +0530

    S_L4PCI_2 test moved to SBSA L4

    - Fix for issues #297 and #317.
    - Moved test_p062 to L4 of SBSA under TEST_ID 601
    - Included all device types in PCIe BDF table.
    - PCIe test rule mapping fixes.
    - PCIe BDF table used to recognise exercisers also.

    Signed-off-by: Srikar Josyula <srikar.josyula@arm.com>
```

Sujana-M, 2023-04-27:
Hi @hrw 

I tried reproducing the issue at my end by adding the PCIe hierarchy as mentioned in https://github.com/hrw/sbsa-ref-status/blob/main/boot-sbsa-ref.sh. I'm facing the below issue while doing so

"qemu-system-aarch64: -device pcie-root-port,id=root_port_1.2,chassis=1,slot=2: MSI-X is not supported by interrupt controller"

I am able to add devices directly to the Host bridge but unable to add a RP. Could you please share if any config was changed at your end to enable the same.

Platform used QEMU 8.0.0 (Downloaded from https://www.qemu.org/download/)  and latest edk2 (https://github.com/tianocore/edk2).

Please Note: The test did not enter a loop before the commit because prior to this commit SBSA was run on RCiEP, RCEC, iEP_EP and iEP_RP. As a result, the test is skipped as in https://github.com/hrw/sbsa-ref-status/blob/main/status-sbsa.yml#L748.  Since the Rule "S_PCIe_02" needs to be checked for all PCIe devices, this test now checks for RP's as well.

Request you to share the logs of QEMU from your end with -v 1 

Thanks,
ACS Team

hrw, 2023-04-28:
Will reply after Linaro Connect and provide requires patches.

hrw, 2023-05-04:
QEMU code/patches: https://github.com/hrw/fork-qemu/releases/tag/hrw-2023.05.04

EDK2 platforms code/patches: https://github.com/hrw/fork-edk2-platforms/tree/hrw-2023.05.04

Those branches have some WIP code to add ITS/SMMU stuff.

hrw, 2023-05-04:
log:
```
Shell> pci
   Seg  Bus  Dev  Func
   ---  ---  ---  ----
    00   00   00    00 ==> Bridge Device - Host/PCI bridge
             Vendor 1B36 Device 0008 Prog Interface 0
    00   00   01    00 ==> Network Controller - Ethernet controller
             Vendor 8086 Device 10D3 Prog Interface 0
    00   00   02    00 ==> Display Controller - VGA/8514 controller
             Vendor 1234 Device 1111 Prog Interface 0
    00   00   03    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   04    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   05    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   06    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   01   00    00 ==> Serial Bus Controllers - USB
             Vendor 1B36 Device 000D Prog Interface 30
    00   02   00    00 ==> Display Controller - Other display controller
             Vendor 1234 Device 1111 Prog Interface 0
    00   03   00    00 ==> Network Controller - Ethernet controller
             Vendor 8086 Device 10D3 Prog Interface 0
    00   04   00    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000E Prog Interface 0
    00   05   08    00 ==> Pre 2.0 device - UNDEFINED
             Vendor 1AF4 Device 1005 Prog Interface 0
Shell> fs0:sbsa.efi -v 1 -l 7

 SBSA Architecture Compliance Suite 
    Version 7.1.1  

 Starting tests for level  7 (Print level is  1)

 Creating Platform Information Tables 
 Using SMC as PSCI conduit
 MADT is at FC0EF598 and length is 1A8 
  Flags 1 
  PE Enabled 1, Online Capable 0
 MPIDR 0 PE num 0 
  Flags 1 
  PE Enabled 1, Online Capable 0
 MPIDR 1 PE num 1 
  Flags 1 
  PE Enabled 1, Online Capable 0
 MPIDR 2 PE num 2 
  Flags 1 
  PE Enabled 1, Online Capable 0
 MPIDR 3 PE num 3 
 PE_INFO: Number of PE detected       :    4 
 MADT is at FC0EF598 and length is 1A8 
 GICH base 2C010000 
 GICH base 2C010000 
 GICH base 2C010000 
 GICH base 2C010000 
 GIC DIS base 40060000 
 GIC RD base Structure 40080000 
 GIC ITS base 44081000 
 GIC ITS ID0 
 GIC_INFO: Number of GICD             :    1 
 GIC_INFO: Number of ITS              :    1 
 GTDT is at FC0EFD18 and length is 84 
 TIMER_INFO: Number of system timers  :    0 
 Watchdog base = 0x50011000 INTID = 0x30 
 WATCHDOG_INFO: Number of Watchdogs   :    1 
PPTT table found at 0x100FC0EF818 with length 0xF0

Cache info * Index 0 *
[... 973 lines cut ...]
       BDF - 0x100
       Link Capabilities reg check failed
       Invalid Link Capabilities 0x1
       Failed on PE -    0 for Level=  7 
       RE_PCI_1
       Checkpoint --  1                  : Result:  FAIL 
       END  

 863 : Slot Cap, Control and Status register rules      
       START  
       Number of bit-field entries to check 26
       IE_REG_4
       Checkpoint --  1                  : Result:  SKIPPED 
       END  

 861 : Check RootPort P&NP Memory Access 
       START  
       BDF - 0x100
       BDF - 0x300
       Received exception of type: 0
       Received exception of type: 0
       Received exception of type: 0
       Received exception of type: 0
       Received exception of type: 0
```

hrw, 2023-05-05:
I think that bug can be somewhere in our QEMU/EDK2 patches.

Sujana-M, 2023-05-09:
Hi @hrw,

Thanks for sharing the QEMU and EDK2 patches. I am still facing the same issue with the edk2-platforms. 

To help further debug on the edk2 patches, I have added few debug prints as part of the test in my fork https://github.com/Sujana-M/sbsa-acs/commit/de9b1f9320828261941d52f10330e818dcccb6a8. Can you please share logs by taking this patch?

Request you to share the logs by running at level 7 and verbosity 1 (-l 7 -v 1).

Thanks,
Sujana

hrw, 2023-05-16:
Applied this commit on top of sbsa-acs:

```
 861 : Check RootPort P&NP Memory Access
       START
       BDF - 0x100
       BDF - 0x200
       BDF - 0x300
    RP BDF - 0x300
    BAR Base - 0x810C5000
       Received exception of type: 0
       Received exception of type: 0
       Received exception of type: 0
```

My command:

```
./code/qemu/build/aarch64-softmmu/qemu-system-aarch64 -machine sbsa-ref
 -m 4096 -smp 4 -cpu max -drive if=pflash,file=SBSA_FLASH0.fd,format=raw
 -drive if=pflash,file=SBSA_FLASH1.fd,format=raw
 -drive file=fat:rw:/home/marcin/devel/linaro/sbsa-qemu/sbsa-ref-status/disks/virtual/,format=raw 
-net nic -net user -usb -device usb-kbd -device usb-tablet -nographic
 -watchdog-action none -no-reboot -monitor telnet::45454,server,nowait
 -serial stdio 
-device pcie-root-port,id=root_port_for_igb,chassis=1,slot=2 -device igb 
-device pcie-root-port,id=root_port_for_ac97,chassis=1,slot=9 -device ac97
```

Sujana-M, 2023-05-19:
Hi @hrw,

I tried the above set of command and still face the below same issue.

qemu-system-aarch64: -device pcie-root-port,id=root_port_for_igb,chassis=1,slot=2: MSI-X is not supported by interrupt controller

hrw, 2023-05-19:
You need forked qemu/edk2-platforms. Links in one of previous comments.

hrw, 2023-05-24:
We track it internally here: https://linaro.atlassian.net/browse/SRCPC-105

hrw, 2023-06-07:
Turned out that this is due to QEMU configuration which disallows 1B/2B access to PCIe. Which is exactly what test 861 does.

We have change for QEMU queued.

chetan-rathore, 2023-06-07:
Thanks @hrw for the update.

As the issue is QEMU configuration related, can we close this?

Thanks,
ACS team

hrw, 2023-08-29:
@chetan-rathore @Sujana-M @gowthamsiddarthd ping

gowthamsiddarthd, 2023-08-29:
Hi @hrw,

SBSA Test 861 corresponds with rule S_PCIe_02. It specifically verifies whether Root Ports (RPs) are capable of facilitating 1-byte (1B) and 2-byte (2B) read and write transactions to and from both Prefetchable and non-prefetchable memory regions. These transactions are directly associated with the memory Base Address Registers (BARs) rather than involving access to the MSI/MSI-X tables. Could you kindly validate the accuracy of this understanding?

Regards,
Gowtham,
The ACS Team

closed 2023-08-31 by hrw