ARM-software/sbsa-acs #328: PCIe tests are not skippable

hrw, 2023-04-24:
I want to run test 861 (related to issue #327) but running "sbsa.efi -l 7 -v 1 -t 861" ends with skipping all PCIe tests:

```
 801 : Check ECAM Presence
       START
       USER OVERRIDE VIA SINGLE TEST - Skip Test
       PCI_IN_01
       Checkpoint --  0                  : Result:  SKIPPED
       END

     *** Skipping remaining PCIE tests ***
```

hrw, 2023-04-24:
val/src/avs_pcie.c has this code:

```c
    status |= p001_entry(num_pe);

    if (status != AVS_STATUS_PASS) {
      val_print(AVS_PRINT_WARN, "\n     *** Skipping remaining PCIE tests *** \n", 0);
      return status;
    }
```

While "-t" argument accepts only one value so it cannot be told "-t 801,861"

gowthamsiddarthd, 2023-04-27:
Hi @hrw,

The ACS team is working on the changes and will raise the PR soon.

Regards,
ACS Team

PR #332 (2023-05-04): Removed limitation on -t and -m options
 - Resolves: #328 
 - Removed the limitation on number of tests and modules to be run using -t and -m options in UEFI and Baremetal.
commit b9ff0565 (2023-05-04): Removed limitation on -t and -m options

 - Resolves: #328
 - Removed the limitation on number of tests and modules
   to be run using -t and -m options in UEFI and Baremetal.
closed 2023-05-04 by chetan-rathore