ARM-software/sbsa-acs #348: Secondary core cannot get test function vector in val_test_entry

foxmmmm, 2023-05-18:
System halts when run **Sbsa.efi** of pre-built v23.01_2.0.0_BETA-0 or built from 27f904f:
[image]
But there is no problem for v22.10_1.1.0 **Sbsa.efi** or run v23.01_2.0.0_BETA-0 **Bsa.efi**.

Tracking this phenomenon, the secondary core (core 7)  does not get correct **index** from val_pe_get_index_mpid(), so the core 7 only gets test function vector from default(0) location: 
[image]

There is a global shared struct g_pe_info_table variable accessed by val_pe_get_index_mpid(), which is initialized by master core with cache-clean-invalidate operation, but secondary cores directly read g_pe_info_table->pe_info->->mpidr/pe_num without cache-invalidate operation in val_pe_get_index_mpid(). General method is done like  mem->data0/mem->data1 in val_set_test_data()/val_get_test_data(): 
val_set_test_data(uint32_t index, uint64_t addr, uint64_t test_data)
{
...
  mem->data0 = addr;
  mem->data1 = test_data;

  val_data_cache_ops_by_va((addr_t)&mem->data0, CLEAN_AND_INVALIDATE);
  val_data_cache_ops_by_va((addr_t)&mem->data1, CLEAN_AND_INVALIDATE);
}
val_get_test_data(uint32_t index, uint64_t *data0, uint64_t *data1)
{
...
  val_data_cache_ops_by_va((addr_t)&mem->data0, INVALIDATE);
  val_data_cache_ops_by_va((addr_t)&mem->data1, INVALIDATE);

  *data0 = mem->data0;
  *data1 = mem->data1;
}

Based on the above method, apply the following patch,
diff --git a/val/src/avs_pe_infra.c b/val/src/avs_pe_infra.c
index 7768912..7a36eba 100644
--- a/val/src/avs_pe_infra.c
+++ b/val/src/avs_pe_infra.c
@@ -177,7 +177,7 @@ val_pe_get_index_mpid(uint64_t mpid)
 
   entry = g_pe_info_table->pe_info;
 
\-  while (i > 0) {
\+  while (i > 0) {val_data_cache_ops_by_va((addr_t)&entry->mpidr, INVALIDATE); val_data_cache_ops_by_va((addr_t)&entry->pe_num, INVALIDATE);
     if (entry->mpidr == mpid) {
       return entry->pe_num;
     }
The system runs fine:
[image]

chetan-rathore, 2023-05-19:
Hi @foxmmmm,

Thanks for the patch and explanation of the issue faced.
We will review and merge it from our end.

Thanks,
ACS team

chetan-rathore, 2023-06-07:
Hi @foxmmmm,

Thanks for sharing the patch, we have raised the changes both in BSA and SBSA repository.
Please also share your review feedback.

Thanks,
ACS team

closed 2023-06-12 by chetan-rathore