ARM-software/sbsa-acs #373: Tests 828/829 (PCIe Power management ones) need better message?

hrw, 2023-07-13:
My emulated platform reports Bochs display card like this:

```
00:02.0 Display controller [0380]: Device [1234:1111] (rev 02)
        Subsystem: Red Hat, Inc. Device [1af4:1100]
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        IOMMU group: 1
        Region 0: Memory at 80000000 (32-bit, prefetchable) [size=16M]
        Region 2: Memory at 8108c000 (32-bit, non-prefetchable) [size=4K]
        Expansion ROM at 81080000 [disabled] [size=32K]
        Capabilities: [80] Express (v2) Root Complex Integrated Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0
                        ExtTag- RBE+ FLReset-
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP- LTR-
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt+ EETLPPrefix+, MaxEETLPPrefixes 4
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
        Kernel driver in use: bochs-drm
        Kernel modules: bochs
```

It is PCI Express card (we had legacy PCI simple VGA before).

Tests 828 and 829 say:

```
 828 : Check Power management cap rules  
       START  
       Number of bit-field entries to check 2
       BDF 0x100 : PASS
       BDF 0x100 : PASS
       PCIe Capability not found for BDF 0x200
       PCIe Capability not found for BDF 0x200
       Failed on PE -    0 for Level=  7 
       RE_REG_2, IE_REG_5
       Checkpoint --  2                  : Result:  FAIL 
       END  

 829 : Check Power management/status rule
       START  
       Number of bit-field entries to check 2
       BDF 0x100 : WARNING Data Select attribute mismatch: 0x1E00 instead of 0x0
       BDF 0x100 : PASS
       PCIe Capability not found for BDF 0x200
       PCIe Capability not found for BDF 0x200
       Failed on PE -    0 for Level=  7 
       RE_REG_2, IE_REG_5
       Checkpoint --  2                  : Result:  FAIL 
       END  
```

Would be nice to see which PCIe Capabilities are not found without going into SBSA ACS source code.

hrw, 2023-07-13:
Test 842 says it better:

```
 842 : Check Power Management rules      
       START  
       BDF - 0x100
       BDF - 0x200
       BDF - 200 does not support Power Management Capability
       Failed on PE -    0 for Level=  7 
       RE_PWR_1
       Checkpoint --  1                  : Result:  FAIL 
       END  
```

chetan-rathore, 2023-08-08:
Hi @hrw,

Changes for the issue are merged.

Thanks,
ACS team

closed 2023-08-08 by chetan-rathore

hrw, 2023-08-17:
```
 828 : Check Power management cap rules  
       START  
       Number of bit-field entries to check 2
       BDF 0x100 : PASS
       BDF 0x100 : PASS
       PCIe Capability 0x1 not found for BDF 0x200
       PCIe Capability 0x1 not found for BDF 0x200
       Failed on PE -    0 for Level=  7 
       RE_REG_2, IE_REG_5
       Checkpoint --  2                  : Result:  FAIL 
       END  
```

thanks