ARM-software/sbsa-acs #388: Why PE test 16 (rule S_MPAM_PE) limits the resource tpye to PE cache only?

StevenHuang0117, 2023-09-08:
Our ARM servers use Neoverse N2 cores, our MPAM supports SLC(system level cache) partitioning. Our firmware reports cache resources type of each MSC node as memory-side cache. But PE test 16 limits the resource tpye to PE cache only. 

1. Our team are confused what exact definitions of PE cache, memory-side cache and LLC here?
2. Whether the PE cache refers to the cache inside the PE, or the cache that actually participates in coherency including SLC, It is called PE cache?
3. SLC is invisible cache, is it still be PE cache or cache participating in consistency?

chetan-rathore, 2023-09-11:
Hi @StevenHuang0117,

The PE test 16 relies on PPTT table to find the LLC node. The last node in a PE hierarchy is treated as an LLC node.

Thanks,
ACS team

StevenHuang0117, 2023-09-12:
Hi @chetan-rathore 

Thanks for your reply！

I still don't understand why LLC node must be PE-cache type as follow codes?
            if (val_mpam_get_info(MPAM_MSC_RSRC_TYPE, msc_index, rsrc_index) ==
                                                                       MPAM_RSRC_TYPE_PE_CACHE) {

Our LLC node is SLC, not L1/L2/L3 cache, so we think we can treat it as memory-side cache, are we wrong?

In addition, I will greatly appreciate you can answer the 3 questions in my first conment. 
As our ARM technical support engineer said, more experts of ARM and other vendors may need to be involved in this discussion to unify defintion of PE cache and memory-side cache.

chetan-rathore, 2023-09-13:
Hello @StevenHuang0117,

We are having internal discussions about the queries raised around the LLC (PE cache or memory side cache).
We will provide a response based on the updates.

Thanks,
ACS team

StevenHuang0117, 2023-10-10:
Hi @chetan-rathore ,
Is there any progress in your internal  discussion?

chetan-rathore, 2023-10-11:
Hello @StevenHuang0117, 

Thanks for your patience.

Here are the updates, please share your inputs also.

-	An SLC can be considered a CPU-side cache if there is only one SCG (System Cache Group) in the system and one NUMA node.
-	An SLC can be considered a memory-side cache if there are multiple SCGs in the system and hence multiple NUMA nodes.
-	Shared caches (for example cluster-caches) may be included in the PPTT when they should be visible to OS SW.

As for the BSA MPAM requirements, it is  only required to apply to a last-level cache (LLC)
-	What is considered the LLC is an IMPDEF system choice
o	For example, a system may consider the cluster-cache as its LLC even if has SLCs.
o	Another example: There is no cluster-cache but there are SLCs that are memory-side caches. In this case the memory-side cache may be the LLC.

The current proposal is to update the test to parse both the PPTT and HMAT tables and introduce a command line argument to obtain whether the System LLC is present in PPTT or HMAT table and accordingly perform check on that cache.

Thanks,
ACS team

StevenHuang0117, 2023-10-17:
Hello @chetan-rathore ,

Thanks for your updates. 

So as your updates, our SLC can be considered a memory-side cache and LLC. 

We think your proposal may fix our issue, so we are looking forward to your updating the code of this test case. Do you have any plan to finish it?

Thanks,
Steven Huang

chetan-rathore, 2023-10-17:
Thanks @StevenHuang0117  for confirming the changes will work for your issue.

We are planning to complete this by next month.

chetan-rathore, 2023-11-29:
Hi @StevenHuang0117,

We have made some changes in the test as discussed above.
Can you please share the debug log with the below image by running the following command  "Sbsa.efi - l 5 -v 1 "
https://github.com/amrathesh/sbsa-acs-fork/suites/18584559547/artifacts/1079759328

You can share the test log directly to the "support-systemready-acs@arm.com" email. 

Please ignore the test status as this is a debug image.

Thanks,
ACS team

chetan-rathore, 2023-12-05:
Hi @StevenHuang0117,

Can you please share update on the debug image run.

Thanks,
ACS team

chetan-rathore, 2023-12-18:
Hello @StevenHuang0117,

Any update on the debug image run.

Thanks,
ACS team

StevenHuang0117, 2023-12-20:
Hi @chetan-rathore 

Sorry for our late test, we have send the test log to the "[support-systemready-acs@arm.com](mailto:support-systemready-acs@arm.com)" email now.

PR #430 (2024-01-30): PE test 16, check for CPOR support for LLC presented as mem-side
 - Added support to parse and check for cpor support by memory side last level cache.
 - Level 1 mem-side cache present near to memory is considered as mem-side LLC.

Fixes: #388
commit 19caa4da (2024-01-30): PE test 16, check for CPOR support for LLC presented as mem-side (#430)

- Added support to parse and check for cpor support by memory
   side last level cache.
 - Level 1 mem-side cache present near to memory is considered
   as mem-side LLC.

Fixes: #388

Change-Id: Id9cdb9cb5f2e87b82418dcc43b04a7fdce88fda6
closed 2024-01-30 by chetan-rathore