ARM-software/sbsa-acs #408: SBSATimer use interrupt issue

hensonwang, 2023-10-19:


chetan-rathore, 2023-10-20:
Hello @hensonwang, 

Can you please provide more details on the issue like
* which test failure is seen
* type of failure (exception, hang)

If you can attach the logs, it will be very helpful.

Thanks,
ACS team

hensonwang, 2023-10-20:
Hi ACS team，
In fact， there is no test item to check SBSA timer LPI interrupt. So I'm not sure if you will add SBSA timer LPI interrupt check item later if BSA spec declare that the system wakeup timer could support LPI，or could you check with your team if the RB_time_08 description is correct? It will be very appreciated if you could help clarify it. Thanks in advance! 

Thanks！

hensonwang, 2023-10-23:
@chetan-rathore ，

Any update for this issue?

Thanks！

rajatgoyal47, 2023-10-26:
Hello @hensonwang,

As per the rule 
"B_TIME_08 On timer expiry, the system wakeup timer must generate an interrupt that must be wired to the GIC as an SPI
or LPI."

You are right as per the rule it is also possible to have the support for LPI interrupt. However, Current version of ACS Tests checks only for the SPI Interrupt. 
We will assess this internally and provide you with an update on the timeline.

Thanks,
ACS team

hensonwang, 2023-10-27:
Thanks for your response. Is this also applied to GTDT timer？ If so, I think GTDT timer definition should be updated based on this change, Do you have any plan to add this part in ACPI spec in the future? 

Thanks!

rajatgoyal47, 2023-10-30:
Hello,

Thank you for providing details on the GTDT Timer discussion.

The BSA/SBSA Specification refers to Generic Timer, and the tests related to timers extract interrupt information from the GTDT ACPI Table. 
I will discuss this with the relevant architects for further clarification and will keep you updated. 

Thank you.

hensonwang, 2023-10-31:
Hi @rajatgoyal47 ，

Thanks for your quick response. waiting for your clarification.

Thanks！

rajatgoyal47, 2023-10-31:
Hi @hensonwang,

Please find the updates below.
 
While the current BSA language permits using an LPI for a System timer, in practice an SPI is the more native signaling method for this type of event that Arm expects.
LPI is less suitable for this type of event due to lack of SW stack support and lack of architectural methods to set the LPI parameters.

Can you please let us know Is there a specific reason why you require using an LPI for the Timer event?

Thanks,
ACS Team

hensonwang, 2023-10-31:
Hi Rajat,

You mean we still can use LPI for GTDT system timer, right? Does Mark Rutland also agree the point?

Thanks！

hensonwang, 2023-11-02:
Hi ACS team，

Another more question related with SBSA timer using LPI. As described by ACPI GT Block Timer structure format in ACPI 6.5, BIOS need report GTx physical timer GSIV, if we use LPI for GTx timer, how could I report the GT Timer? Waiting  for your reply.

Thanks！

rajatgoyal47, 2023-11-02:
Hi @hensonwang,

Although the BSA specification currently mentions LPI for System timer, we recommend against using LPI. 
Currently there are no plans to update the ACPI tables to accommodate the LPI option. 

Additionally,  we are considering a change in the BSA specification language regarding the use of LPI. 

It would be helpful if you can let us know Is there a specific reason why you require using an LPI for the Timer event?

Thanks,
ACS Team

prime-zeng, 2023-11-14:
@rajatgoyal47  

> Hi @hensonwang,
>
> Although the BSA specification currently mentions LPI for System timer, we recommend against using LPI. Currently there are no plans to update the ACPI tables to accommodate the LPI [...]
1.  We have a system which the system timer is using the LPI   🙂， this is the top reason
2.  Some tech reason from architecture view of point:
(1) In a huge Soc system, the SPI resource is expensive, we are trying to save such kind of resources.
(2) If the system has got the TEE enabled, usually we need to use the secure system timer as the secure time source,  and then if we have only one system timer as SPI, we have to using LPI timer as the non-secure (linux kernel) timer source.  

And as we understand , there is no  known reason for non-use of  LPI timer, could you help to explain the consideration from your side?

Thanks.
> Thanks, ACS Team

chetan-rathore, 2023-11-23:
Hello @hensonwang,

Thanks for explaining the use of LPI in your case.

While SPI is a limited resource, when considering system wakeup timers then there are only 2 global system interrupts to consider (secure wakeup and non-secure wakeup) so allocating an SPI for each should not be too onerous. LPIs are meant to solve an interrupt scaling problem where many assignable devices might ask for their own IRQ resources.
We are currently looking into deprecating the use of LPIs for peripherals that don’t expose an architected method to program the LPI parameters. Accordingly, we’d suggest considering using SPI for each of the wakeup timers.

Thanks,
ACS team

hensonwang, 2023-11-23:
Hi @chetan-rathore，

Thanks for your reply，I'll discuss this internally and let you know if we have any more questions. Thanks for your detail explanation.

chetan-rathore, 2023-11-30:
Hi @hensonwang ,

Please reopen the ticket if any further support needed.

Thanks,
ACS team

closed 2023-11-30 by chetan-rathore