ARM-software/sbsa-acs #409: Enhancement to val_pcie_mem_get_offset

robbiek-xsl, 2023-10-30:
A useful enhancement to val_pcie_mem_get_offset would be to provide additional parameters to better describe the intended use of the offset returned.

Parameters:
1. BDF - Indicates the PCIe device to be accessed using the returned offset
2. bar index - Indicates particular BAR region within the device
3. purpose - indicate if the test intends to do a read only vs a read/modify/write
4. size - size of the access to be done by the test

Many PCIe devices have BAR regions where certain memory accesses have side effects, for instance a write to a particular address can trigger a doorbell.  When a design contains a mixture of various PCIe devices, there may not be a single offset value that can satisfy all the devices in the system.

chetan-rathore, 2023-11-16:
Hello @robbiek-xsl,

Thanks for the inputs, we also internally had a discussion around the 804 and 805 tests logic and will be making some changes to the tests.
More details captured here: https://github.com/ARM-software/bsa-acs/issues/224

Thanks,
ACS team

chetan-rathore, 2023-12-18:
Hello @robbiek-xsl,

Required changes for the issue are merged in the ACS.

Thanks,
ACS team

closed 2024-01-18 by chetan-rathore