ARM-software/sbsa-acs #418: Wakeup timer interrupt usage issue

hensonwang, 2023-12-13:


hensonwang, 2023-12-13:
This discussion is based on issue #408.
I saw description - It is permissible to build a system with no support for SPI in page 24 in Arm® Base System Architecture 1.0C document, because the SPI quantity is very limited，if we want to use LPI for wakeup timer, seems it is reasonable, right? Did your team ever think about such non-SPI system design solution as well as system wakeup timer? 
Waiting for your reply.

Thanks！

hensonwang, 2024-01-08:
Hi ACS team,

Any update for this question?

Thanks！

chetan-rathore, 2024-01-11:
Hello @hensonwang,

We are discussing this internally and will share the updates by next week.

Thanks,
ACS team

hensonwang, 2024-01-12:
Hi chetan，

Great, waiting for your reply.

Thanks！

chetan-rathore, 2024-01-18:
Hi @hensonwang,

While the rules permit the use of LPI, in practice there is a lack of support for LPI for these IPs in the corresponding FW/ACPI tables and in standard OS drivers. Moreover, there are issues with using LPIs for these such as lack of an architected method to set the LPI parameters and programming interfaces that were designed for level-sensitive interrupts”

BSA specification will be updated to remove LPI as an option for UART, WD, and Timer.

Thanks,
ACS team

hensonwang, 2024-01-22:
Hi chetan，

Got it. could you let me know when will you update the BSA spec?

Thanks!

chetan-rathore, 2024-01-25:
Hello @hensonwang,

Can you please redirect your query regarding specification update to support-systemready-acs@arm.com

Thanks,
ACS team

hensonwang, 2024-01-26:
Hi Chetan，

Thanks for your reminder, How can I redirect this issue to your specification team? Do you have any detail instruction so I can I do that?

Thanks!

hensonwang, 2024-02-20:
Hi Chetan，

Thanks for your support on this issue ticket, Can close it now.

Thanks！