ARM-software/sbsa-acs #437: PCIe Test 025 : The Enable No Snoop Bit(bit11) value and attribute is incorrect

jumpmanhou, 2024-03-13:
From PCIe Spec: the Enable No Snoop bit is RW and default value is 1,  but the test_p025_data.h set this bit Read_ONLY and value 0 as expected result, which is mismatch with PCIe Spec.

gowthamsiddarthd, 2024-04-08:
Hi @jumpmanhou,

This test specifically targets RCEC devices, not RCiEPs, in accordance with the PCIe specification.

According to the PCIe specification:
RCiEP: This bit enables/disables HW’s ability to set non-cacheable attribute for transactions.
RCEC: Hardwired to 0.

During the ACS testing, it verifies whether the RCEC device maintains this bit value as 0, effectively making the attribute read-only due to its hardcoded nature. Thus, there appears to be no deviation from the PCIe specification in this scenario.

Regards,
Gowtham,
ACS Team

jumpmanhou, 2024-04-08:
Hi Gowtham，
Thanks for clarifying the test case. I do make a mistake for the target device. But I still have questions about  test025:
1. for RCiEP, 
  bit[9] Phantom Functions Enable, I Hardwired to 0 for this bit, So I suppose the Attribute should be READ_ONLY instead of READ_WRITE 
  bit[10] Aux power PM Enable is the same,  I Hardwired to 0 for this bit, I suppose the Attribute should be READ_ONLY instead of STICKY_RW

2. For RCEC, bit[11] Enable No Snoop, I didn't find the requirement in PCIe Specification that RCEC  MUST hardwired this bit to 0. Now I just set this bit as READ_WRITE with default value to 1.  So can you point the detailed chapter/page for the RCEC requirement?

jumpmanhou, 2024-04-09:
Thanks, Gowtham. I will hardwire ENS to 0. But I think it should be a recommended value instead of a MUST requirement. The requirement is from BSA Specification but I can't find the same requirement from PCIe Specication.

closed 2024-04-09 by jumpmanhou