ARM-software/sbsa-acs #439: Features not supported causes test item to fail

hysen0523, 2024-03-20:
Test item "6. Check Cryptographic extensions" failed. The root cause is that bits 7:4, 11:8, 15:12 of ID_AA64ISAR0_EL1 are all zeros. Should this test case be waived? Or, update this test case to get a "skip" result if FEAT_AES/FEAT_SHA1/FEAT_SHA256 not supported? Or other solutions? Thanks a lot.

[image]

chetan-rathore, 2024-03-21:
Hi @hysen0523,

The tests mentioned above has recently been updated, Can you confirm if the results are with latest binary or with SR or ES systemready image.
 
Also can you provide information on the SBSA level selected to run the tests.

Thanks,
ACS team

chetan-rathore, 2024-04-03:
Hello @hysen0523,

You are correct that the rules tested by the mentioned test have dependency on export restrictions.
We will add a user command line parameter to get if export restrictions allowed or not, and test SKIP or FAIL will depend on the flag value.

Thanks,
ACS team

closed 2024-04-08 by chetan-rathore