ARM-software/sbsa-acs #446: WORKSPACE can point to any place

hrw, 2024-04-04:
I noticed that some code from SBSA ACS is now in BSA ACS. And while BSA ACS builds fine, SBSA ACS stopped building (gcc command split to show all arguments in easier to read way):

```
Building ... /home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/uefi_app/SbsaAvs.inf [AARCH64]
make[1]: Wejście do katalogu '/home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/uefi_app/SbsaAvs'
"aarch64-linux-gnu-gcc" 
-MMD 
-MF /home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/uefi_app/SbsaAvs/OUTPUT/SbsaAvsMain.obj.deps  
-g 
-Os 
-fshort-wchar 
-fno-builtin 
-fno-strict-aliasing 
-Wall 
-Werror 
-Wno-array-bounds 
-include AutoGen.h 
-fno-common 
-ffunction-sections 
-fdata-sections 
-DSTRING_ARRAY_NAME=SbsaStrings 
-g 
-Os 
-fshort-wchar 
-fno-builtin 
-fno-strict-aliasing 
-Wall 
-Werror 
-Wno-array-bounds 
-include AutoGen.h 
-fno-common 
-mlittle-endian 
-fno-short-enums 
-fverbose-asm 
-funsigned-char 
-ffunction-sections 
-fdata-sections 
-Wno-address 
-fno-asynchronous-unwind-tables 
-fno-unwind-tables 
-fno-pic 
-fno-pie 
-ffixed-x18 
-mcmodel=small 
-O0 
-O0 
-I/home/marcin/devel/linaro/sbsa-qemu/code/ShellPkg/Application/bsa-acs/val/ 
-I/home/marcin/devel/linaro/sbsa-qemu/code/ShellPkg/Application/bsa-acs/val/sbsa 
-I/home/marcin/devel/linaro/sbsa-qemu/code/ShellPkg/Application/bsa-acs/val/common 
-I/home/marcin/devel/linaro/sbsa-qemu/code/ShellPkg/Application/bsa-acs/ 
-D DISABLE_NEW_DEPRECATED_INTERFACES 
-c 
-o /home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/uefi_app/SbsaAvs/OUTPUT/./SbsaAvsMain.obj 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/test_pool/ete/operating_system 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/test_pool/ras/operating_system 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/test_pool/pmu/operating_system 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/test_pool/mpam/operating_system 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/test_pool/exerciser/operating_system 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/test_pool/memory_map/operating_system 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/test_pool/smmu/operating_system 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/test_pool/pcie/operating_system 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/test_pool/watchdog/operating_system 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/test_pool/gic/operating_system 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/test_pool/pe/operating_system 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/uefi_app 
-I/home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/uefi_app/SbsaAvs/DEBUG 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2-libc/StdLib 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2-libc/StdLib/Include 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2-libc/StdLib/Include/Aarch64 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/EmbeddedPkg 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/EmbeddedPkg/Include 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/EmbeddedPkg/Test/Mock/Include 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/MdePkg 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/MdePkg/Include 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/MdePkg/Test/UnitTest/Include 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/MdePkg/Test/Mock/Include 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/MdePkg/Library/MipiSysTLib/mipisyst/library/include 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/MdePkg/Include/AArch64 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/MdeModulePkg 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/MdeModulePkg/Test/Mock/Include 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/MdeModulePkg/Include 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg 
-I/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Include 
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/uefi_app/SbsaAvsMain.c
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sbsa-acs/uefi_app/SbsaAvsMain.c:26:10: fatal error: val/sbsa/include/sbsa_val_interface.h: Nie ma takiego pliku ani katalogu
   26 | #include "val/sbsa/include/sbsa_val_interface.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [GNUmakefile:1146: /home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/uefi_app/SbsaAvs/OUTPUT/SbsaAvsMain.obj] Błąd 1
make[1]: Opuszczenie katalogu '/home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sbsa-acs/uefi_app/SbsaAvs'
```

I looked at INF files and noticed that you assume that WORKSPACE variable points to edk2 checkout (again split to multiple lines):

```
  GCC:*_*_*_CC_FLAGS   = -O0 -I${WORKSPACE}/ShellPkg/Application/bsa-acs/val/
 -I${WORKSPACE}/ShellPkg/Application/bsa-acs/val/sbsa
 -I${WORKSPACE}/ShellPkg/Application/bsa-acs/val/common
 -I${WORKSPACE}/ShellPkg/Application/bsa-acs/
```

Can it get done in a way which does not requires WORKSPACE to be set to edk2/ checkout?

SrikarJosyula, 2024-04-22:
Hi Marcin,

We are trying to get to know more about the issue.
WORKSPACE is set to $PWD by edksetup.sh which we source before building SBSA-ACS with edk2. And as far as I know, you can point EDK_TOOLS_PATH first and then source edksetup.sh for WORKSPACE to not point to edk2 checkout.

If that is the scenario, you might get an error similar to this while sourcing acsbuild.sh-
"build.py...
: error 000E: File/directory not found in workspace
ShellPkg/Application/sbsa-acs/uefi_app/SbsaAvs.inf is not found in packages path:
/home/user/run/edk2/edk2-libc"

Can we know how you're managing to build without WORKSPACE pointing to edk2? We have made related changes locally but wanted to know more so that we could provide a solution apt for the issue.

Thanks,
ACS Team.

hrw, 2024-04-22:
```
$ cd ~/devel/linaro/sbsa-qemu/code/
$ cat setup.sh
#!/bin/bash

export WORKSPACE=$PWD
export GCC_AARCH64_PREFIX=aarch64-linux-gnu-
export GCC49_AARCH64_PREFIX=aarch64-linux-gnu-
export PACKAGES_PATH=$WORKSPACE/edk2:$WORKSPACE/edk2-platforms:$WORKSPACE/edk2-non-osi:$WORKSPACE/edk2-libc

make -C edk2/BaseTools -j10
. edk2/edksetup.sh

$ . setup.sh
[.. lot of output ..]

$ make sbsa-acs bsa-acs
```

section of makefile (ignore "Linux-x86_64" PATH part - there is no such dir):
```
sbsa-acs:
        PATH=$$PWD/edk2/BaseTools/Bin/Linux-x86_64:$$PWD/edk2/BaseTools/BinWrappers/PosixLike:$$PATH \
        WORKSPACE=$$PWD \
        GCC_AARCH64_PREFIX=aarch64-linux-gnu- \
        PACKAGES_PATH=$$WORKSPACE/edk2:$$WORKSPACE/edk2-platforms:$$WORKSPACE/edk2-non-osi:$$WORKSPACE/edk2-libc \
        . edk2/ShellPkg/Application/sbsa-acs/tools/scripts/acsbuild.sh
        cp Build/Shell/DEBUG_GCC49/AARCH64/Sbsa.efi ../disks/virtual/Sbsa.efi

bsa-acs:
        PATH=$$PWD/edk2/BaseTools/Bin/Linux-x86_64:$$PWD/edk2/BaseTools/BinWrappers/PosixLike:$$PATH \
        PATH=$$PWD/edk2/BaseTools/Bin/Linux-x86_64:$$PWD/edk2/BaseTools/BinWrappers/PosixLike:$$PATH \
        WORKSPACE=$$PWD \
        GCC_AARCH64_PREFIX=aarch64-linux-gnu- \
        PACKAGES_PATH=$$WORKSPACE/edk2:$$WORKSPACE/edk2-platforms:$$WORKSPACE/edk2-non-osi:$$WORKSPACE/edk2-libc \
        . edk2/ShellPkg/Application/bsa-acs/tools/scripts/acsbuild.sh
        cp Build/Shell/DEBUG_GCC49/AARCH64/Bsa.efi ../disks/virtual/Bsa.efi
```

hrw, 2024-04-22:
[my whole makefile](https://github.com/ARM-software/sbsa-acs/files/15069000/makefile.txt)

I use it to build QEMU, Trusted Firmware (for sbsa-ref and virt), EDK2 (sbsa-ref and virt), SBSA/BSA ACS, EFI Shell, my ArmCpuInfo etc..

With debug/release/noopt, gcc/clang etc.

hrw, 2024-04-22:
```
$ ls -1
arm-acs
armcpuinfo
Build
edk2
edk2-libc
edk2-non-osi
edk2-platforms
makefile
qemu
scripts
setup.sh
trusted-firmware-a
```

arm-acs/ holds bsa-acs and sbsa-acs directories which are symlinked to edk2/ShellPkg/Application/

SrikarJosyula, 2024-04-23:
Hi Marcin,

Could you please try the SBSA-ACS build on your end with the changes in the fork? Do let us know if the build went fine.
https://github.com/SrikarJosyula/sbsa-acs/tree/master
https://github.com/SrikarJosyula/bsa-acs/tree/main

Thanks,
ACS team

hrw, 2024-04-23:
Both bsa-acs and sbsa-acs built fine.

PR #452 (2024-04-23): Removing WORKSPACE dependency from .inf files
1.  Resolves #446 
2.  -I compiler option now uses BSA_PATH instead of WORKSPACE.
3.  BSA_PATH to be set by acsbuild.sh script.
commit 110a79c7 (2024-04-23): Removing WORKSPACE dependency from .inf files (#452)

- Resolves #446
 - -I compiler option now uses BSA_PATH instead of WORKSPACE.
 - BSA_PATH to be set by acsbuild.sh script.
closed 2024-04-23 by chetan-rathore