ARM-software/sbsa-acs #454: PE Test 16 (Check MPAM LLC Requirements) doesn't support the case where ACPI MAPM's MSCs are PCC based

sunnywang-arm, 2024-04-29:
When the MSCs are PCC based, the Base Address field of the MPAM MSC node is the subspace ID of the PCC channel rather than the Base address of the memory-mapped MPAM register space. In this case, we’ll need to get the MPAM register space from the ACPI PCCT table. However, sbsa-acs doesn’t support this case.

samerhaj, 2024-04-29:
MPAM over "ACPI PCC" is described in [MPAM Firmware-backed (Fb) Profile Specification ](https://developer.arm.com/documentation/den0144/0100/?lang=en )

chetan-rathore, 2024-05-03:
Hi @samerhaj  and @sunnywang-arm,
 
The MPAM test will need update for the cases when MPAM register space is exposed via PCCT table.
We are working on it and will keep you updated.

Thanks,
ACS team

PR #486 (2024-10-04): Add support for PCCT based MSC
 - allocated memory for PCC info table and calling
   val_pcc_create_info_table.
 - allocated memory for pcc info table.
 
 Fixes: #454 
 Deps: https://github.com/ARM-software/bsa-acs/pull/344
closed 2024-10-04 by chetan-rathore