ARM-software/sbsa-acs #460: Drop PCIe tests

hrw, 2024-05-08:
PCI Express tests are present in BSA ACS and none of them has SBSA tags attached.

So we have duplicated tests to run, they have different names and tags attached.

gowthamsiddarthd, 2024-05-22:
Hi @hrw,

While it may seem that these tests are duplicates, it is important to note that they serve distinct purposes by targeting different device categories.

The BSA tests are specifically designed to verify the functionality of Endpoint and Root Port devices. These tests ensure that EPs and RPs conform to BSA specifications and standards.

On the other hand, the SBSA tests have a broader scope and are intended to check various types of integrated devices. These include:

- Integrated Endpoints (iEPs)
- Root Complex Integrated Endpoints (RCiEPs)
- Root Complex Event Collectors (RCECs)

If you have any further questions or need additional clarification, please feel free to ask.

Regards,
Gowtham
ACS Team

hrw, 2024-05-23:
Thanks.

Will look into which types of device categories I have in my testing setup to make sure all needs are covered.

closed 2024-05-23 by hrw