ARM-software/sbsa-acs #468: Duplicate test case 302 in testcase checklist

anncheng-arm, 2024-07-17:
https://github.com/ARM-software/sbsa-acs/blob/9fc2359604ba9ee6a57bb84ee5bc79585f183bb1/docs/arm_sbsa_testcase_checklist.rst?plain=1#L140

rajatgoyal47, 2024-07-25:
Hello Ann,

Currently we have written the testcase_checklist based on the combination of RULE_ID + Level.

In this particular case [(Test 302)](https://github.com/ARM-software/sbsa-acs/blob/master/test_pool/smmu/operating_system/test_i002.c) covers the following 

- S_L5SM_01, S_L5SM_02 - Checks for SMMUv3.2 or Higher for systems with Level >= 5 and Level < 8
- S_L8SM_01 - Checks for SMMUv3.3 or higher for SBSA Level >= 8

Hope this helps.

Regards,
Rajat

anncheng-arm, 2024-07-29:
I see, thanks a  lot for the information.

closed 2024-07-29 by anncheng-arm